wled/WLED #1034: additional connection of warm white LEDs

Basti149, 2020-07-08:
Ich nutze seit gestern diese Software und bin total zufrieden es hat alles geklappt. Nun wollte ich zu den WS2812 LEDs noch ein Strip Warm Weiße LEDs anschließen. Leider werden diese nicht unterstützt. Es wäre super wenn es eine Möglichkeit geben würde wenn man Warm Weiße LED Strip noch mit Ansteuern könnte. Zureichen würde es  auch wenn es ein Schieberegler mit Helligkeitsanpassung geben würde oder einfach ein Ein/ Aus Schalter für Warm Weiße LEds.
Ich hoffe meine Programm vorschlag wird umgesetzt. 

I have been using this software since yesterday and I am totally satisfied that everything worked out fine. Now I wanted to connect a Strip Warm White LED to the WS2812 LEDs. Unfortunately these are not supported. It would be great if there would be a possibility to control the Warm White LED Strip. It would also be good if there was a slider with brightness adjustment or just an on/off switch for Warm White LEds.
I hope my program suggestion will be implemented. 

Translated with www.DeepL.com/Translator (free version)

huggy-d1, 2020-07-08:
I believe you want to use analog led strip using wled. This is an option but requires custom code configuration followed by compiling firmware that only works for analog led strips. 

It may be possible there is a version for analog LEDs already compiled for you esp8266 platform. You will need to check the firmware builds.

Basti149, 2020-07-09:
> I believe you want to use analog led strip using wled. This is an option but requires custom code configuration followed by compiling firmware that only works for analog led strips.
>
> It may be [...]

Danke für die schnelle Antwort.
Gut zu wissen das es auch eine Version für 4 pin RGB Leds gibt. Da muss ich später mal gucken ob es eine fertige bin schon gibt.

Mir geht es eigentlich darum ich habe ein Regal über meiner Couch auf diesen sind WS2812 LEDs verbaut.  Die bereits über Wled laufen. Unter diesen Regal wollte ich ein warm weißes LEDs Strip anbauen als so zusagen als "Lese" Beleuchtung. Ich hatte mir Vorgestellt das es ein weiteren Freien GPIO gibt mit den man zusätlich einfachen Warm Weiße LED Strip  zu den WS2812 Strips an den EsP8266 anschließen könnte ohne ein weiteren Controller zu benutzen. Mir würde es reichen wenn es ein virtuellen Taster in der WLED Oberfläche geben würde mit den man ein GPIO Ein/Ausschalten kann. Andere Variante ist ein Schieberegler mit der man die Helligkeit einstellen kann. Dies ist aber kein muss. Es würde auch Ein/Aus reichen.

Thanks for the quick reply. Good to know that there is also a version for 4 pin RGB leds. Since I have to look later times if there is a ready am already. I'm actually about I have a shelf installed above my couch on these are WS2812 LEDs. They're already running through Wled. Under these shelves I wanted to add a warm white LED strip as a "read" illumination. I had imagined that there is another Free GPIO with the additional easy Warm White LED Strip to connect the WS2812 Strips to the EsP8266 without using another controller. It would be enough for me if there was a virtual button in the WLED surface with which you can switch a GPIO on/off. Another variant is a slider with which you can adjust the brightness. But this is not a must. It would also be enough on/off.

huggy-d1, 2020-07-13:
If you are really set on using a single ESP8266, you can find an addressable warm white (WW) LED strip to mount under the shelf.  The data line into the WW addressable strip can be run from the same pin as the RGB strip.  The effect on the RGB strip would be replicated for any effect that handles the WW after the ESP8266 is configured to drive a RGBW strip.  The RGB and W would all come out the same data pin.  My guess is this sounds good in theory but would not work at all in practice.
I see addressable WW LED strips are available though (one ex: https://www.adafruit.com/product/2435 ).
I recommend reviewing options on Quindor's LED site (with details on driving ANALOG LED strips): https://quinled.info/2018/09/17/rgbwwcw-led-strips/
All manner of LED knowledge can be found on the same site, starting at https://quinled.info

Basti149, 2020-07-15:
Thank you for your help. The easiest is if you could add an exit in Wled with which you could control the Warm White Strips over a transistor (Dimmen). There would also be a normal switching outlet with which I could easily switch the warm white strips e. g. B over a relay. Maybe you can integrate something like that into the firmware.

Danke für ihre Hilfe. Das einfachste ist ja wenn man in Wled ein Ausgang hinzufügen könnte mit den man die Warm Weißen Strips über ein Transistor steuern könnte (Dimmen). Es würde auch ein normaler Schaltausgang reichen mit den ich die warm weißen  Strips einfach schalten könnte z.B über ein Relais.
Vielleicht kann man so etwas in die Firmware integrieren.

ubergeekseven, 2020-11-30:
@Basti149 that is exactly what I am looking for. The firmware, I recently discovered after years of custom coding, is amazing, fast and reliable.
Having the ability to output through another gpio to trigger pwm on that white strip would make this firmware perfect. 

I have under cabinet lighting that's finished this way just using pwm output to a mosfet with a port expander that I use as sections throughout the room. This way I can use 1 controller to reduce WiFi noise and address hundreds of strands with one output. 

I could live with the button input being able to turn into an output with pwm. I have not yet used the input. I haven't dug into the code at all yet. Just discovered the firmware this last week and have been blown away with the simplicity of setup. Maybe it's not possible with what's happening in the code but, if it is, that would be my only want in this setup. Pwm is definitely needed with it.

maximecb, 2021-01-04:
It would be cool if WLED supported analog RGB/RGBW/solid-color strips out of the box. There are a lot of analog RGB strips in the world. They cost a lot less than addressable strips, and not every application needs individually applicable LEDs. Supporting these "singe-pixel" strips also opens the door to controlling devices like spotlights or other kinds of lighting in a room (anything you can drive with PWM) directly from the WLED app.

I found out about WLED because I was writing my own firmware for addressable LEDs and I was just wondering if someone had already gone through the effort of building the firmware and a convenient app. I am glad that someone did. However, I'd really like to be able to control other (simpler) lighting devices besides addressable strips from the WLED app. The implementation complexity required to add support for PWM and "one-pixel" devices seems not too difficult and really worth doing. It would help WLED cement itself as a dominant player in home automation.

LeducH, 2021-01-05:
> The implementation complexity required to add support for PWM and "one-pixel" devices seems not too difficult and really worth doing. 

sure that would be the simplest way to do this. Now since usual smart LED require a data line the analog only das two lines: the positive and negative line. To use the  `D4` pin on the ESP8266 some controller must go in between. 

Where would I start to dig into this topic?

Basti149, 2021-01-18:
> @Basti149 that is exactly what I am looking for. The firmware, I recently discovered after years of custom coding, is amazing, fast and reliable.
> Having the ability to output through another gpio to [...]

That's exactly what I would be interested in. Unfortunately, I do not know how the integration works.

Basti149, 2021-01-18:
> It would be cool if WLED supported analog RGB/RGBW/solid-color strips out of the box. There are a lot of analog RGB strips in the world. They cost a lot less than addressable strips, and not every [...]

Exactly you could also connect the digital LEDs with the normal RGB Leds.

Basti149, 2021-01-18:
> 
> 
> > The implementation complexity required to add support for PWM and "one-pixel" devices seems not too difficult and really worth doing.
> 
> sure that would be the simplest way to do this. Now since usual smart LED require a data line the analog white strips only has two lines: the positive and negative line (RGB four lines). To use the `D4` pin on the ESP8266 some controller must go in between.
> 
> Where would I start to dig into this topic?

Either you use WS2812 LEDS or someone would extend the software to integrate the normal RGB LEDS. What is also very interesting for me.

ubergeekseven, 2021-02-09:
> sure that would be the simplest way to do this. Now since usual smart LED require a data line the analog white strips only has two lines: the positive and negative line (RGB four lines). To use the [...]

I have used multiplexer chips to expand ports out and still have pwm over them. Those use i2c. That would allow multiple outputs that are pwm.

if wanting to use only one output and be tolerant of the 12 volt white strips, i use a mosfet to control the brightness with pwm out.

maximecb, 2021-02-09:
@ubergeekseven That's cool. Any chance you could list the chips you used for this, and maybe sketch a little circuit diagram? Could be useful for everyone who wants to use WLED with some PWM LEDs.

ubergeekseven, 2021-02-10:
> @ubergeekseven That's cool. Any chance you could list the chips you used for this, and maybe sketch a little circuit diagram? Could be useful for everyone who wants to use WLED with some PWM LEDs.

To be clear, I have not modified the WLED setup for this use. I have used it in many other setups. 

for multiplexing I use the MCP23017 chip. It gives you 16 I/O and you can expand that out to several more chips for more I/O. I have used that chip for actuating garden water solenoids as well as adding control for multiple lighting projects. Since there are not a lot of pins available on esp devices, this is perfect for expanding functionality. It also allows me to create large HID macro keyboards for things like controlling my CNC. Adafruit has an easy to use library for it as well.

The chip I use for bidirectional voltage conversion is the SN74ahct125N if needing to control with 3.3 output and the end device needs 5v. However, this I only use to control neopixels through a raspi so the output can trigger reliably and not destroy the pin or board. I also like to use this with relays from 3.3v devices.

The mosfet I use for driving the PWM for my 12v white lighting needs to be fast switching. I use the RFP30N06LE 60V 30A Nchannel mosfet for this purpose. They are potentially a bit overkill but, I use them in multiple applications and the original ones I installed for my cabinet lighting are going strong almost two years in. I have not had a single issue with these. You can use them to trigger each line of an RGB strip as well. I no longer use straight RGB when I have easy access to addressable LEDs though.
They are simple enough to wire up. when looking straight on at it with the label towards you. The far right pin is connected to power ground. The middle pin goes to the ground of the LED strip. The left pin goes to the pin used for PWM output from the microcontroller. The positive output from the power supply runs to the LED strip input. The mosfet switches the ground connection. 
You can drive DC motors the same way as long as you are within specs of the mosfet and you can always add a heat sink to the tab of the mosfet to help dissipate heat if needed. 
I suppose that some mosfets may have a different wiring setup package. I use the through hole type and pout them in headers to swap out if needed.

Here is a diagram that shows what I am talking about: [Adafruit](https://www.google.com/url?sa=i&url=https%3A%2F%2Flearn.adafruit.com%2Frgb-led-strips%2Fusage&psig=AOvVaw3whSJeRWLPvbPbS5BaLiep&ust=1613046238945000&source=images&cd=vfe&ved=0CAIQjRxqFwoTCID9w9Kn3-4CFQAAAAAdAAAAABA7)

[image]

Basti149, 2021-03-01:
> To be clear, I have not modified the WLED setup for this use. I have used it in many other setups.
>
> for multiplexing I use the MCP23017 chip. It gives you 16 I/O and you can expand that out to [...]

If you run the MCP23017 chip with 5V or 3V. I am currently using this chip for my Tasmota variant. Now the question is you absolutely need a Mosfet or does it work without it. For about 6 RGB LEDs per channel.

ubergeekseven, 2021-03-09:
@Basti149 good question. I'm not sure if the pin voltage tolerances on that multiplexer. Is have to look it up on the sheet but, with the cost of that chip being much more than a mosfet that can handle lots of current. It's not a bad easy to protect it. I have not tried that because my strips tend to be much longer. 
I've run a small ws2812 ring from the output of a pi without issue before but, I've had odd colors come through too. Sometimes the length of wire alone is enough to mess with the signal. It's with a shot though. The chip in quantities of 10 tend to be reasonable.

blazoncek, 2022-07-25:
CCT (request from OP) is supported since November 2021 #2285

closed 2022-07-25 by blazoncek