wled/WLED #1036: WLED on Wemos D1 Mini losing WiFi

iansime, 2020-07-09:
I have loaded the latest WLED binary onto a Wemos D1 Mini and connected it to my home WiFi. I am able to connect to the board from my Android phone (WiFi) or Desktop PC (wired) but soon after the WiFi drops out for a while then eventually comes back.

I put a port scanner on my network and I can see Port 80 regularly not responding, even though th PING to the board never drops.

I tried loading the previous binary (using ESPHome-Flasher-1.2.0-windows-x64.exe) and upgrading to the newest version from there but no change in behaviour.

As a last step I uploaded a sample sketch to the board (simple web server) and that never drops the WiFi connection.

Any tips or troubleshooting steps I can follow?

JohCo-hassio, 2020-07-09:
I also have version 0.10.0 for some time now
And he indeed loses his connection after some time with, for example, Home assistant. 
This is simply accessible via the Wled app,

iansime, 2020-07-10:
I have just installed "Sound Reactive WLED Version 0.10.1" as a comparison and so far that seems rock solid.
Not a single connection loss so far after an hour or so of solid use.

gastonMM, 2020-07-24:
where do you download version 0.10.1?

iansime, 2020-07-24:
https://github.com/atuline/WLED/releases/tag/0.10.1

pbolduc, 2020-08-31:
Is this a duplicate of #1025 ?

closed 2021-01-08 by stale[bot]