wled/WLED #1037: WLED WiFi fails when /win&RD=0 command sent to it

GarSys, 2020-07-10:
**Describe the bug**
When I execute:
    action: call-service
    service: rest_command.hyperion_tv_off 
the wifi connection of the Wemos D1 Mini crashes and the only way to bring it back online is to power cycle the Wemos D1 Mini.  This happens whether it is in override mode (from the web interface) or not.

**To Reproduce**
In Home Assistant:
```
rest_command:
  hyperion_tv_on:
    url: 'http://192.168.17.159/win&RD=1'
  hyperion_tv_off:
    url: 'http://192.168.17.159/win&RD=0'

sensors.yaml:
- platform: rest
  name: Hyperion TV
  resource: http://192.168.17.159/json/info
  value_template: '{{ value_json.live }}'
  force_update: true
  device_class: illuminance

```
SIMPLEST WAY TO SEE THE PROBLEM:
Just send the http://192.168.17.158/win&RD=0 command via your browser.  The WiFi connection will then fail.

**Expected behavior**
The goal is to be able to change the source (WLED or Hyperion) that drives the LEDs (no crash when /win&RD=0 is sent).  Also to get feedback from WLED as to which state the ".live" parameter is (thus keeping Home Assistant current with WLED).

**WLED version**
 - Board: Wemos D1 Mini, (and ESP32 dev)
 - Version: "Namigai", Build 2005030
 - Environment: esp8266 2_6_3 (1)
 - Format:  Binaries from main GitHub site

**Additional context**
I have swapped out the Wemos Di Mini with an esp32 board and with a nodemcu (8266) board with the same sad results.

Thank you for your help!

jes1417, 2020-07-10:
I use this in my HA switch.yaml and it works perfectly for turning off and on live data from Hyperion.

````
  - platform: rest
    name: WLED Hyperion Toggle
    resource: http://192.168.1.193/json/state
    body_on: '{"lor": 0}'
    body_off: '{"lor": 1}'
    is_on_template: '{{value_json.lor == 0}}'
    headers:
      Content-Type: application/json
````

GarSys, 2020-07-10:
Sweet, I will try it out.
May I then suggest that on "https://github.com/Aircoookie/WLED/wiki/HTTP-request-API" that the information on "&RD=" be removed? :)

sansillusion, 2020-07-10:
> I use this in my HA switch.yaml and it works perfectly for turning off and on live data from Hyperion.
>
> ```
>   - platform: rest
>     name: WLED Hyperion Toggle
>     resource: [...]

Could you give me some pointers ?
I have HA running on a raspberry pi4, I have Hyperion running and configured in HA and the Wled instance. But I don't have a switch.yaml and if I create one with the content you provided (changing ip for my wled instance) nothing changes. I can't find how to work with it or even if creating a switch.yaml file made a difference. I would like to be able to disable Hyperion and control my wled instance from HA.

GarSys, 2020-07-11:
@sansillusion "switch.yaml" is used when you have in your configuration.yaml file the line: "switch: !include switch.yaml".  If you have not split your configuration into individual files, then where you keep you "switches:" is where you place the code.  It will create an entity named switch.wled_hyperion_toggle for your use.

sansillusion, 2020-07-11:
@GarSys Thanks ! That worked like a charm !

Aircoookie, 2020-07-14:
Need yet to confirm this, but I believe disabling `RD` and then receiving realtime data before rebooting WLED could indeed cause issues because it might fill up the receive buffer. JSON `lor` and HTTP api `LO=1` (the latter of which is part of latest master source but not of 0.10.0) work by still accepting, but ignoring the data.
Now that we have the live override feature, it would probably be easiest to deprecate/remove `RD` especially as it is causing issues.

closed 2020-11-19 by stale[bot]