wled/WLED #1038: ps property in json API always -1

kareem613, 2020-07-10:
**Describe the bug**
The JSON API always return -1 for the `ps` value regardless of the currently applied preset.

**To Reproduce**
1. Make a GET request to http://{wled_host}/json/state
2. note the value of the ps property is -1. 
3. Apply preset #1 via the web ui or API
4. Make the same GET request from step 1 and note the ps value is still -1. 

**Expected behavior**
The ps value should be 1.
Note. Preset 1 is used as an example but this doesn't work for any preset.

**WLED version**
 - Board: Wemos D1
 - Version 0.10.0
 - Format Binary

**Additional context**
None

Thank you for your help!

huggy-d1, 2020-07-13:
If you configure preset #2 and then assign preset #2, does the json api respond with -1, or 2?

huggy-d1, 2020-07-13:
What happens when index < 16 and version = 2 or 3?  Nothing??!!

```
  if (index < 16) {
    if (ver != 1) return false;
    <do stuff for                                         index < 16 & ver = 1>
  } else {                                                     index = 16
    if (ver != 2 && ver != 3) return false;
    <do stuff for                                         index = 16 & ver 2 or 3>
    if (ver == 2) { //versions before 2004230 did not have opacity
      <do stuff for                                       index = 16 & ver 2>
    }
    <do stuff for                                         index = 16 ver 2 or 3>
  }
  currentPreset = index;
  isPreset = true;
  return true;
```
Missing { } around "return false" is causing the else to refer to index < 16.

Aircoookie, 2020-07-14:
Could not reproduce this... When I apply a preset that has something saved to it, `ps` contains that preset number both in 0.10.0 and latest master.
Note that right now the current state is not compared against presets, which means that if anything (color or even just brightness) is modified, `ps` reverts to `-1`. In an upcoming presets update, I will change the behavior so that the value of `ps` ignores brightness changes by default. Is there perhaps some automation which sets a value after you applied the preset?

@huggy-d1 That's right, nothing happens. Versions 2&3 are the preset format containing all segments, which is only supported in preset 16 because of EEPROM space constraints (all other presets being only 20 bytes and thus only saving one segment)

huggy-d1, 2020-07-14:
Is it possible the EEPROM value for ver is != 1?  Would that cause the problem?

Aircoookie, 2020-07-14:
If the ver was != 1, the preset wouldn't be applied as no other version but 1 is valid for slots 1-15... If the preset visibly gets applied, also the `ps` should have the correct value. I don't think I know how it could fail. @kareem613 did you have multiple segments while reproducing this issue?

huggy-d1, 2020-07-14:
What if.. OTA done over top of a non WLED installation?

closed 2020-11-19 by stale[bot]