wled/WLED #1041: Buttons with assigned presets

Partizan7676, 2020-07-13:
Will be very useful if we can assign presets to buttons with time option to switch back or not. Also it can be used as a contact sensor with HA for example.

axlan, 2020-07-16:
I made a usermod that has this sort of functionality: https://github.com/Aircoookie/WLED/tree/master/usermods/blynk_relay_control . Uses Blynk instead of GPIO, but it would be pretty straight forward to modify and to add additional buttons.

I think the challenge in making this an included feature that there's a lot of configuration involved in interpreting every possible GPIO input (which pin, bouncing, active high/low, duration of press, etc.), along with a lot of possible actions that could be triggered.

Partizan7676, 2020-07-17:
@axlan thanks for your attention! but in my case i need contact sensor/ may be i'll try usermod for pir sensor

Aircoookie, 2020-07-18:
I agree! WLED will have multi button support in the future, then you'll be able to assign up to three presets to each button (single/double/long press). About the timer feature that switches back i'm not yet sure, but that has also been requested multiple times and may get added :)

Partizan7676, 2020-07-19:
@Aircoookie it is great! But better and easy implement single press for preset. Timer for switch back to last set - will also good idea.

josephlmanning, 2020-07-24:
> I agree! WLED will have multi button support in the future, then you'll be able to assign up to three presets to each button (single/double/long press). About the timer feature that switches back i'm not yet sure, but that has also been requested multiple times and may get added :)

Consider this one more official request!  I'd really like to be able to toggle from preset 1 to preset 2 on button press, hold preset 2 for x seconds, and then after x seconds toggle back to preset 1.

blazoncek, 2022-02-21:
Multi button support is available since 0.13-b2.
https://github.com/Aircoookie/WLED/pull/1977
Closing.

closed 2022-02-21 by blazoncek