wled/WLED #1042: WLED BULID

kevinhuihui, 2020-07-14:
**Take a look at the wiki and FAQ, perhaps your question is already answered!**
[FAQ](https://github.com/Aircoookie/WLED/wiki/FAQ)
**### Hi Aircoookie!
when i was found this wonderful and amazing source, i was shocked!
what a great WLED!   it's lighted my life!**

_i really really put my myself into this project! find and dig...._

also after i modified my personal effects wants add to WLED,(LATER im gonna to post...) i met a problem stoppped me while buliding(im using vscode and Platform IO(PIO EXTENSION) files..... lots of warnings... could you pls give me some suggestions?

also i tried bulid in arduino IDE, many times , it works but...u know...speed took very long(here also a problem met in Arduino IDE while buliding...wants to bulid sucess except delete “arduino” folder from (WLED-0.10.0\WLED-0.10.0\wled00\src\dependencies\**arduino**）im not sure whether u know or not?

any suggetsions about my question i appreciate it!

thanks! great day!

> Such as following(from vs code):

`Compiling .pio\build\nodemcuv2\src\alexa.cpp.o
Compiling .pio\build\nodemcuv2\src\blynk.cpp.o
Compiling .pio\build\nodemcuv2\src\button.cpp.o
Compiling .pio\build\nodemcuv2\src\colors.cpp.o
In file included from src\fcn_declare.h:79:0,
                 from src\wled.h:84,
                 from src\alexa.cpp:1:
src\FX.h:41:0: warning: "MIN" redefined [enabled by default]
 #define MIN(a,b) ((a)<(b)?(a):(b))
 ^
In file included from C:\Users\user\Documents\Arduino\libraries\FastLED/platforms/esp/8266/fastled_esp8266.h:5:0,
                 from C:\Users\user\Documents\Arduino\libraries\FastLED/platforms.h:34,
                 from C:\Users\user\Documents\Arduino\libraries\FastLED/FastLED.h:52,
                 from src\FX.h:34,
                 from src\fcn_declare.h:79,
                 from src\wled.h:84,
                 from src\alexa.cpp:1:
C:\Users\user\Documents\Arduino\libraries\FastLED/platforms/esp/8266/clockless_block_esp8266.h:8:0: note: this is the location of the previous definition
 #define MIN(X,Y) (((X)<(Y)) ? (X):(Y))
 ^
In file included from src\fcn_declare.h:79:0,
                 from src\wled.h:84,
                 from src\blynk.cpp:1:
src\FX.h:41:0: warning: "MIN" redefined [enabled by default]
 #define MIN(a,b) ((a)<(b)?(a):(b))
 ^
In file included from C:\Users\user\Documents\Arduino\libraries\FastLED/platforms/esp/8266/fastled_esp8266.h:5:0,
                 from C:\Users\user\Documents\Arduino\libraries\FastLED/platforms.h:34,
                 from C:\Users\user\Documents\Arduino\libraries\FastLED/FastLED.h:52,
                 from src\FX.h:34,
                 from src\fcn_declare.h:79,
                 from src\wled.h:84,
                 from src\blynk.cpp:1:
C:\Users\user\Documents\Arduino\libraries\FastLED/platforms/esp/8266/clockless_block_esp8266.h:8:0: note: this is the location of the previous definition
 #define MIN(X,Y) (((X)<(Y)) ? (X):(Y))
 ^
In file included from src\fcn_declare.h:79:0,
                 from src\wled.h:84,
                 from src\button.cpp:1:
src\FX.h:41:0: warning: "MIN" redefined [enabled by default]
 #define MIN(a,b) ((a)<(b)?(a):(b))
 ^
In file included from C:\Users\user\Documents\Arduino\libraries\FastLED/platforms/esp/8266/fastled_esp8266.h:5:0,
                 from C:\Users\user\Documents\Arduino\libraries\FastLED/platforms.h:34,
                 from C:\Users\user\Documents\Arduino\libraries\FastLED/FastLED.h:52,
                 from src\FX.h:34,
                 from src\fcn_declare.h:79,
                 from src\wled.h:84,
                 from src\button.cpp:1:
C:\Users\user\Documents\Arduino\libraries\FastLED/platforms/esp/8266/clockless_block_esp8266.h:8:0: note: this is the location of the previous definition
 #define MIN(X,Y) (((X)<(Y)) ? (X):(Y))
 ^
[... 340 lines cut ...]
src\NpbWrapper.h:271:12: warning: enumeration value 'NeoPixelType_End' not handled in switch [-Wswitch] 
src\NpbWrapper.h: In member function 'void NeoPixelWrapper::SetBrightness(byte)':
src\NpbWrapper.h:290:12: warning: enumeration value 'NeoPixelType_None' not handled in switch [-Wswitch]
     switch (_type) {
            ^
src\NpbWrapper.h:290:12: warning: enumeration value 'NeoPixelType_End' not handled in switch [-Wswitch] 
src\NpbWrapper.h: In member function 'RgbwColor NeoPixelWrapper::GetPixelColorRgbw(uint16_t) const':    
src\NpbWrapper.h:300:12: warning: enumeration value 'NeoPixelType_None' not handled in switch [-Wswitch]
     switch (_type) {
            ^
*** [.pio\build\nodemcuv2\src\alexa.cpp.o] Error 1
src\NpbWrapper.h:300:12: warning: enumeration value 'NeoPixelType_End' not handled in switch [-Wswitch] 
src\NpbWrapper.h: In member function 'uint8_t* NeoPixelWrapper::GetPixels()':
src\NpbWrapper.h:309:12: warning: enumeration value 'NeoPixelType_None' not handled in switch [-Wswitch]
     switch (_type) {
            ^
src\NpbWrapper.h:309:12: warning: enumeration value 'NeoPixelType_End' not handled in switch [-Wswitch] 
src\NpbWrapper.h: In member function 'void NeoPixelWrapper::cleanup()':
src\NpbWrapper.h:326:12: warning: enumeration value 'NeoPixelType_None' not handled in switch [-Wswitch]
     switch (_type) {
            ^
src\NpbWrapper.h:326:12: warning: enumeration value 'NeoPixelType_End' not handled in switch [-Wswitch]
*** [.pio\build\nodemcuv2\src\colors.cpp.o] Error 1
*** [.pio\build\nodemcuv2\src\button.cpp.o] Error 1
===================================== [FAILED] Took 14.15 seconds =====================================`

Aircoookie, 2020-07-14:
Hi,
I am very happy that you like WLED!
In Arduino IDE, for me a full build typically takes 90 seconds while re-building with only a few bits changed takes about 12 seconds. This might vary depending on your system of course.
About the error in PIO, it seems to bbe referencing the Adafruit CC3000 library, but that is not actually a dependency of WLED as far as I know. What are the modifications you made and did the unmodified version compile?

huggy-d1, 2020-07-14:
Were you ever able to compile WLED?  Then you made changes and suddenly you cannot?   Uncomment (or use git to go back in time) chunks of added code to identify what new library is causing so much conflict.

kevinhuihui, 2020-07-15:
> Hi,
> I am very happy that you like WLED!
> In Arduino IDE, for me a full build typically takes 90 seconds while re-building with only a few bits changed takes about 12 seconds. This might vary [...]

> Very thanks for ur response! 

i've found conflict which is ur methioned....i installed that library long long time ago.. when i was compiling i selected use arduino library as reference in PIO, so casued this problem.... but now! see below... hhah... life is starting...

Creating BIN file ".pio\build\nodemcuv2\firmware.bin" using ".pio\build\nodemcuv2\firmware.elf"
=================================== [SUCCESS] Took 176.46 seconds ===================================

Environment            Status    Duration
---------------------  --------  ------------
nodemcuv2              SUCCESS   00:02:56.460
esp01                  IGNORED
esp01_1m_ota           IGNORED
esp01_1m_full          IGNORED
esp07                  IGNORED
d1_mini                IGNORED
heltec_wifi_kit_8      IGNORED
h803wf                 IGNORED
esp32dev               IGNORED
esp8285_4CH_MagicHome  IGNORED
esp8285_4CH_H801       IGNORED
esp8285_5CH_H801       IGNORED
d1_mini_5CH_Shojo_PCB  IGNORED
d1_mini_debug          IGNORED
d1_mini_ota            IGNORED
custom_LEDPIN_4        IGNORED
custom_LEDPIN_16       IGNORED
custom_APA102          IGNORED
custom_WS2801          IGNORED
custom32_LEDPIN_16     IGNORED
wemos_shield_esp32     IGNORED
m5atom                 IGNORED
travis_esp8266         IGNORED
travis_esp32           IGNORED
==================================== 1 succeeded in 00:02:56.460 ====================================

kevinhuihui, 2020-07-15:
> Were you ever able to compile WLED? Then you made changes and suddenly you cannot? Uncomment (or use git to go back in time) chunks of added code to identify what new library is causing so much conflict.

Thanks a lot! have a great day!

toto79, 2020-07-16:
you can give https://gitpod.io#https://github.com/Aircoookie/WLED/tree/master an try

run 1st: pip3 install -U platformio
run 2nd: platformio run -e <env:type> , e.g. platformio run -e d1_mini or platformio run -e esp01_1m_full (as defined in platformio.ini)

find your firmware.bin file in folder .pio/build/<env:type>, save as... and flash to your esp
good summary you'll find also at https://tasmota.github.io/docs/Gitpod/

closed 2020-08-04 by Aircoookie