wled/WLED #1044: Feature Request: FX: Random Pixel left

macxsm, 2020-07-16:
Hi,
one nice FX preset would be: some random blink over all LEDs and slightly turn off one after another LED until only one is left that stays on for at least some seconds or until the next preset is selected or a button pressed. I could think of many other random selection presets not to mention here. 
The background: I have a kind of "mini" bar and want to let random decide which whiskey to taste next :-D
I was trying to program this in another framework but my programming skills are too poor for this (maybe I should drink less and program more but this is only the second option ;-) )

Ciao!
Marco

Aircoookie, 2020-07-18:
Hi!
I like the idea of your proposed effect a lot! Another similarly nice idea would be a slot machine/wheel of fortune type effect where a single LED is quickly moving across the strip, gradually getting slower and then stopping at a random position!

macxsm, 2020-07-20:
That would be great, too!