wled/WLED #1046: Enable custom 16 cells with effects in turn. With a radio port 433Мhz!

Nikolay16700011, 2020-07-18:
Good afternoon. Recently there was an idea to make control of WLED by illumination, using a remote control of 433MHz. The phone is not always convenient. I have a question. How would I switch the custom 16 slots of favorite effects? In what code does this happen? I'm not good at programming, if you can tell me how to implement it. I did a change of effects++, but if you only like 10 favorites, flipping through more than 100 effects in a row is a sad task. I want to use the radio remote button to switch my favorite 16 slots. Do you have any ideas on how to make it easier? Thanks!

Nikolay16700011, 2020-07-18:
Found is apply the Preset(). I think I did something wrong with the shutdown function. Everything is working now. Maybe you can improve something? Tell me.  Bit of code:

#include <RCSwitch.h>

RCSwitch mySwitch = RCSwitch();
int nomer;
long previousMillis = 0;

unsigned long receivedCode;
//Use userVar0 and userVar1 (API calls &U0=,&U1=, uint16_t)

//gets called once at boot. Do all initialization that doesn't depend on network here
void userSetup()
{ mySwitch.enableReceive(5);  // приемник на D1 ESP (DATA) 
  
}

//gets called every time WiFi is (re-)connected. Initialize own network interfaces here
void userConnected()
{

}

//loop. You can use "if (WLED_CONNECTED)" to check for successful connection
void userLoop()
{
  // if(receivedCode>0) {Serial.print( receivedCode);} //раскомментировать, чтобы посмотреть код кнопки пульта, в мониторе порта.

  receivedCode=mySwitch.getReceivedValue(); 
 if(receivedCode==16736113) {if(millis() -previousMillis >500)  // если надо добавить пульт. То добавляем  if(receivedCode== новый код!!!) 
  { previousMillis = millis();
     toggleOnOff();                                            // ON/OFF
    colorUpdated(NOTIFIER_CALL_MODE_BUTTON);
  } else {
    applyMacro(macroButton);
  } 
}
   if(receivedCode==16736114) {if(millis() -previousMillis >500)  
  { previousMillis = millis(); nomer+=1; applyPreset(nomer);
                    colorUpdated(NOTIFIER_CALL_MODE_FX_CHANGED); if(nomer>16||nomer==0)nomer=1;}  
/*effectCurrent += 1; if (effectCurrent >= MODE_COUNT) effectCurrent = 0; переключения эффектов друг за другом */
 
   }
mySwitch.resetAvailable();  
}

Aircoookie, 2020-07-18:
Hi!
What shutdown function? On first glance everything seems good to me :)
One small thing: `if(nomer>16||nomer==0)nomer=1;}` should be before `applyPreset(nomer);` because otherwise it will try to apply the invalid preset 17 (and also skip over preset 1)

Nikolay16700011, 2020-07-19:
Thank you I found this error: "if(number>16||number==0)number=1;} should be earlier, apply Preset(number);". I'm always confused about the placement. 
Yesterday I experimented, everything works. But the remote response is delayed, and the delay is different. I don't know if this is how the "RC Switch.h"library works. I have no experience with 433MHz radio at all. My module wl 101-341, and remote 4 button 433MHz. I was looking for a ready-made example to get the code from the remote. Maybe another library or radio module will work better? As long as everything suits me, there is a switch, brightness adjustment, and a selection of favorite effects. Left the first 4 effects to choose from. 16 effects for a long time to rewind with one button.

//### CODE:

#include "wled.h"
#include <RCSwitch.h>

RCSwitch mySwitch = RCSwitch();
int nomer;
long previousMillis = 0;
int svet;
unsigned long receivedCode;

void setup() { 

mySwitch.enableReceive(5);  // приемник на D1 ESP (DATA) 
  
WLED::instance().setup();

  
}

void loop() {WLED::instance().loop();

//if(receivedCode>0) {Serial.print( receivedCode);} //раскомментировать, чтобы посмотреть код кнопки пульта в мониторе порта.

  
  receivedCode=mySwitch.getReceivedValue();   // включение светодиодной ленты
 if(receivedCode==16736113) {if(millis() -previousMillis >200)  /* если надо добавить пульт. То добавляем 
if(receivedCode== новый код!!!) */
  { previousMillis = millis();
     toggleOnOff();                                            // ON/OFF
    colorUpdated(NOTIFIER_CALL_MODE_FX_CHANGED); // обновление ленты
  } 

}
   if(receivedCode==16736114) {if(millis() -previousMillis >200)  // переключение избранных эффектов
  { previousMillis = millis(); nomer+=1;
  if(nomer>4||nomer==0)nomer=1; applyPreset(nomer);colorUpdated(NOTIFIER_CALL_MODE_FX_CHANGED);}  /*effectCurrent += 1; if (effectCurrent >= MODE_COUNT) effectCurrent = 0; переключения эффектов друг за другом */
 //Serial.print(nomer);
   }

if(receivedCode==16736120) {if(millis() -previousMillis >200)                                                                   // регулировка света
  { previousMillis = millis(); svet = svet+50; if(svet>280){svet=5;}; bri = svet; colorUpdated(NOTIFIER_CALL_MODE_FX_CHANGED);  // регулировка света шаг+50,min 5
// Serial.print(bri);
   }
}
  // changeBrightness(-10); 

mySwitch.resetAvailable();

 
}

Aircoookie, 2020-07-19:
Hi, I don't have experience with 433MHz either :)
Maybe the delay is caused by the colorUpdated() function not actually applying it instantly though.
You could try replacing `NOTIFIER_CALL_MODE_FX_CHANGED` with `NOTIFIER_CALL_MODE_BUTTON`, maybe it helps!

Nikolay16700011, 2020-07-19:
Thank you to the author of "WLED" for a great project. I would also like to play color music from the line input, or microphone. That would be cool. I'm thinking about how to do it.

Aircoookie, 2020-07-19:
Take a look at https://github.com/atuline/WLED :)

Nikolay16700011, 2020-07-19:
The light show is already Yes. And why I don't see it on YouTube reviews. How do I use it?

closed 2020-07-20 by Nikolay16700011