wled/WLED #1047: Very good with presets after the esp reboot

karen8000, 2020-07-19:
Good afternoon friend 🙃😉I ask for a sling for my English 

Very does not swell the mode of saved preset, namely all saved time and wicket of presets but after the reboot esp8266 01s with a memory of 4mb all have to a new tune. Even if the presets active button 🔳 gpi0 does not work 😔 until you turn off the presets. 
I would like to see the function of blinking to the rhythm of the song if it is possible to realize the 🙃

WLED is a great substitute for Chinese controllers who are forced to use applications and here everything with the web server is very coo.

Aircoookie, 2020-08-03:
Hi, awesome that you like WLED!

I do not understand the issue clearly, sorry. Are your saved presets gone
after a reboot? ESP8266 01s usually only have 1MB of memory, you might have
more luck using a binary with `1m` in the name
For sound reactivity, check out
https://github.com/atuline/WLED

karen8000, 2020-08-04:
Friend, I mean, auto scrolling presets and immeno settings after the reboot with
A esp 01s I have with 4mb memory firmware stands for 4mb full. 

- [ 

> Привет, классно, что тебе нравится WLED! Я не понимаю вопроса ясно, извините. Ваши сохраненные пресеты исчезли после перезагрузки? ESP8266 01s обычно имеют только 1 Мбайт памяти, вы можете иметь больше удачи , используя двоичный файл с `1m` во имя Для звука реактивностью, проверьте https://github.com/atuline/WLED

Friend, I mean, auto scrolling presets and immeno settings after the reboot with
A esp 01s I have with 4mb memory firmware stands for 4mb full. 

- [ 
[image]

closed 2022-02-21 by blazoncek