wled/WLED #1049: truncate spaces after ssid

hamster65, 2020-07-19:
I am new to wled. I flashed the esp and connected to the AP with an iphone. I configured wifi several times but it would not connect and revert to AP mode. It turned out that there was a space character added after the ssid. Maybe that is something to mention in the FAQ or maybe we could truncate it.

Aircoookie, 2020-07-19:
Hi and welcome!
Do you believe the extra space has been added by the phone (autocorrect) or was it there previously?
We cannot truncate as an SSID ending in a space is valid (although the question is who would do that), but adding a warning if a space is the first or last character is probably a good idea!

hamster65, 2020-07-21:
Hi, I tried to reproduce it several times and there was no space... I did not erase / reflash the esp, however.

garyd9, 2020-07-26:
You wouldn't believe what is valid in a SSID:  It's defined as 32 bytes.  That's it.  32 nul (0x00) bytes is a valid SSID.  While the wording the standard changes on occassion, as of 2012 it was "0-32 octets with an unspecified or UTF-8 encoding"

hamster65, 2020-07-26:
wow. I guess there are tons of devices that cannot handle that. After all, some even have problems with special characters in the wifi password.

JojoS62, 2022-06-08:
I ran into the same trouble, maybe I did not delete the preset 'Your_Network' not correctly and hat a trailing space, which is not visible in the WebGUI. Otherwise, I also don't know how to get a trailing space, but it was also saved Firefoxes passwords and so had the problem again in the next trials.
The ESP could not connect to network and started AP again, as intended. Then I tried to set the ssid/pw with the web installer, and that worked. Of course, because I entered the correct ssid here.
But as the trailing space is valid, it would be not ok trim the input and it would break compatiblity. This is issue is stale, so I recommend also to close it. A more user friendly solution would be a list with available access points to pick a name as it is. The name needs to be transfered to a manual input field, manual entry needs to be also possible for hidden ssids.

richard-scott, 2022-12-09:
I've just had this on [v0.13.3](https://github.com/Aircoookie/WLED/releases/tag/v0.13.3), likewise, I don't know if I didn't delete "Your\_Network" correctly, or if my phone added it in.