wled/WLED #1050: DMX Settings Page missing

callumb, 2020-07-20:
So I have managed to (eventually) compile WLED to include DMX however there is no settings button to configure it. going to IP/settings/dmx shows a blank page and going to IP/dmxmap gives the map so its complied correctly. 

Any advice would be appreciated.

jbucky86, 2020-07-26:
I had the same issue. The getV function is not in settings_dmx.htm

jwingefeld, 2020-07-29:
I can confirm this. Will fix it as soon as i get to it. Until then i recommend compiling an older version. 0.10.0 seemed to work. But i will check this more in depth.

Aircoookie, 2020-08-03:
Seems like the issue was introduced with the NPM html autogeneration script not respecting the `WLED_ENABLE_DMX` define.
Thank you @jbucky86 for the fix in #1070, but I do not like that it includes the setting page twice in `html_settings.h`, so I investigated further where the root issue lies.

The server will actually look for `%DMXMENU%` in the html and replace it with the HTML for the DMX menu button if DMX out support is enabled. The problem is that this was only in `html_settings.h` and not in the `settings.htm` file (where it arguably doesn't belong since it is not valid HTTP). I will update the script to automatically add `%DMXMENU%` again after the UI settings button :)

Aircoookie, 2020-08-03:
Should be fixed now!

closed 2020-08-03 by Aircoookie

callumb, 2020-08-03:
Amazing! Thanks everyone for looking at this! :) I will got away and test just now!