wled/WLED #1052: RGB order per segment

s1nbad, 2020-07-22:
I have a 4 x LED WS2811 rings and have wired them up in series to give a total of 89 pixels. ( (37+24+16+12) 
It seems that some are RGB and others are GRB so... when I choose Green 65 are green 24 are red, Red produces 65 red & 24 green, blues are fine.  Apart from this color issue all else works.

Is it possible to use the segment feature to get around this issue ?

huggy-d1, 2020-07-23:
If you are just setting a single color across the entire strip (with or without an effect), you could use segmenting to set the appropriate color in each segment to get the same color along the entire pieced together setup.
If you are doing effects with multiple colors, you can change the colors in each segment for the effects to pull it off.

bjornbergenheim, 2020-07-28:
There is work for the next version to support multiple pins for leds. I think they will solve this with using fastled, and the implementation I've seen you add each led string with RGB order. So by using multiple pins, you should be able to use different RGB-order on the strings.

closed 2020-12-04 by stale[bot]

Matoran, 2021-01-16:
> There is work for the next version to support multiple pins for leds. I think they will solve this with using fastled, and the implementation I've seen you add each led string with RGB order. So by using multiple pins, you should be able to use different RGB-order on the strings.

Hi, I have the same problem with led strips(150leds/strip and 5 strips). My first 3 strips have an order and the 2 last strips have another. Fun fact it's from the same supplier on aliexpress :D. I only use one signal pin for everything.

bjornbergenheim, 2021-01-16:
Honestly, I would send those back and ask for a refund (or just ask for refund, you rarely send stuff back with ali). Or were they separate purchases?

Matoran, 2021-01-16:
Separate purchases with the same store, I'm in touch with the vendor. He's telling me that every led strip is GRB they do not do RGB strips. Will do some more checks on my side but I'm 99% sure is another RGB order.

jaaaa-aaaack, 2021-01-20:
> 
> Found something: Can use COLOR_ORDER_OVERRIDE flag with other parameters to override this section :)
> 
> https://github.com/Aircoookie/WLED/blob/75c219d6c6200cac0d3363f92e5369d19ad9cd28/wled00/NpbWrapper.h#L46

I've just discovered that a system I've built goes back and forth between some RGB and GRBs, the COO flag looks like it could solve it!

Does it allow for a series of non-sequential LEDs to have their colour order overidden, like 11-20, 25-27 etc?

Matoran, 2021-01-20:
No actual system cannot override non-sequential LEDs.
But you can edit the code and adapt it to your config
https://github.com/Aircoookie/WLED/blob/02663702181c2b395d04e16d7e4eb83cbe6699cc/wled00/NpbWrapper.h#L327-L329
replace the previous code by something custom
```C
if (indexPixel >= 50 && indexPixel < 100) co = COL_ORDER_GRB; 
if (indexPixel >= 150 && indexPixel < 200) co = COL_ORDER_RGB; 
```
https://github.com/Aircoookie/WLED/blob/02663702181c2b395d04e16d7e4eb83cbe6699cc/wled00/const.h#L109-L114