wled/WLED #1053: Ability to have hardwired momentary buttons/switches to toggle WLED or active presets

SamPhoenix91, 2020-07-22:
**Is your feature request related to a problem? Please describe.**
Sometimes API or WebUI is not the fastest way to turn off WLED - you are next to the lights and want them off.

**Describe the solution you'd like**
A settings page to be able to configure input pins to toggle the lights or activate presets/effects

**Describe alternatives you've considered**
Another microcontroller with ESPHome to active Home Assistant to control WLED

**Additional context**
Add any other context or screenshots about the feature request here.

Thank you for your ideas for making WLED better!

camcs1, 2020-07-31:
This is already supported I think. Checkout the 'Time & Macros' page in settings 

https://github.com/Aircoookie/WLED/wiki/Macros 

https://github.com/Aircoookie/WLED/wiki/HTTP-request-API

Aircoookie, 2020-08-03:
You can connect a normally open pushbutton between GPIO 0 and ground! With the links by @camcs1 you can customize the button behavior but it does toggle on/off by default.

jaydenrussell, 2020-08-11:
> You can connect a normally open pushbutton between GPIO 0 and ground! With the links by @camcs1 you can customize the button behavior but it does toggle on/off by default.

Exactly what I was able to do. Finding this info about using a normally open pushbutton and what GPIO to use. Once setup there is a good guide for the marcros for the button actions.

Maybe a hardware guide just for an switch.

blazoncek, 2021-05-28:
This feature request has been enhanced in recent development to include multiple buttons, switches and analogue input.
It is already in PR state.

LankyGit1, 2021-07-26:
Hi @blazoncek I saw your comment above. I am using a cupboard entry switch with latest 0.13.0-b2 and wondered if you can help.
I am using an 8266. The issue is that the function is inverted. All the docs I found were that a switch should pull the input low. Rather than me having to redo the wiring I wondered if an inverted switch option could be created. I would suggest that the default switch function was input low being switch preset on and high being switch preset off. 

I have led preferences-> Button 0 set on the switch option.
Time & macro -> Button action short set to 1. 
Also preset 1 being the light function I want.

Maybe I have just got something configured incorrectly, any help greatly appreciated.

Thanks
Paul

blazoncek, 2021-07-26:
Just reverse the presets in button assignment and you are done.
BTW join WLED Discord for better support.

LankyGit1, 2021-07-26:
@blazoncek if I understand you correctly. The first screenshot works (but inverse functionality) but the 2nd only switches on and then never off.
[image]
[image]

blazoncek, 2021-07-26:
0 is not a preset.
Use two presete, one for On and second for Off.
Look into wiki for how-to.

LankyGit1, 2021-07-26:
Thanks set preset 2 as off which I had guessed preset of 0 was.
Looked at wikis but they were out of date and you have to admit that is not the most intuitive for a new user. I still suggest that you set the default action for switch to be the other way round ie on being a 0v input as that is what all the docs say to wire. But beggars cannot be choosers, thanks again for your time. 
Might have a go at editing the wiki as spent ages this afternoon trying to find the right info.

blazoncek, 2021-07-27:
Updated Wiki a bit.