wled/WLED #1054: Improved Alexa  support

halfajobj, 2020-07-22:
Hi there I have recently moved from google to alexa, i can control my wled device via alexa however i cant change the effect via voice ie rainbow etc. Could a wled skill be created for this or implemented in next version

Aircoookie, 2020-08-03:
Hi,
since WLED uses an emulation of the local Philips Hue api (which does not have a concept of effects), this is unfortunately not directly possible.
It would be possible to develop a skill for WLED with this functionality, but I neither have the resources or cloud infrastructure to implement it. Maybe someone else does though :)

siggel, 2022-09-18:
might solve this: https://github.com/Aircoookie/WLED/pull/2787

closed 2022-10-25 by Aircoookie