wled/WLED #1055: How to add an effect

Nikolay16700011, 2020-07-23:
Good afternoon. Tell me how you can add your effect to the list. To display them in the WLED app. And how to summon it. I want to add two additional points: 1. I want to add an analog clock effect with arrows. 2. Make the output ESP =1. The number of effects will increase to 113.

fishbone-git, 2020-07-23:
try this:
https://github.com/Aircoookie/WLED/wiki/Add-own-functionality

I iwll assume that you have already downloaded the code and successfully built code from the unmodified source.

Nikolay16700011, 2020-07-24:
> попробуйте это:
> https://github.com/Aircoookie/WLED/wiki/Add-own-functionality
> 
> Я предполагаю, что вы уже загрузили репозиторий и успешно создали бинарный файл из неизмененного источника.

closed 2020-07-24 by Nikolay16700011