wled/WLED #1060: AP mode webserver stops responding on multiple connections

Adityakush, 2020-07-24:
**Describe the bug**
Hi. I use WLED in AP mode with chip esp12F. And I found out that if multiple clients connect to the AP simultaneously or even if a single clients connects and then disconnects the WIFI and connects again, the heap keeps decreasing which is shown in the info tab and after about  2 or 3 times of reconnecting to the AP, webserver doesnt respond back after that though the AP is still there, though it still works as normal with an IR remote at this time also. It takes a power cycle fro the webserver to respond back. I also tried to compile from source and found that changing from lwip v1.4HigherB/w to v2.0LowerMemory changes this response a bit. Upon reconnecting about 3 or 4 times, it simply restarts the WLED instead of webser not responding. I dont know much but I think that a part of heap is allotted upon a new connection to the AP and even if the same client disconnects the WIFI and connects again then a new part of heap is allotted to it again which is not freed if free heap drops very low and eventually either webserver stops responding or wled restarts.
Is there a workaround for that. Please help

**To Reproduce**
Use wled in AP mode. Connect to the AP, open webserver then disconnect AP, connect again about 3-4 times.

**Expected behavior**
If same clients disconnects from AP and connects again, the part of the heap alotted to it earlier should be freed and hence the problem of no free heap should not arrive and if multiple clients connect to the AP simultaneously then a limit should be set depending on free heap left.

**WLED version*10.1*
 - Board: ESP-12F
 - Format Both Pre compiled Binary and Self Compiled

Thank you for your help!

Adityakush, 2020-07-24:
Hi !
I found out just now that connecting to the AP and disconnecting and then connecting the same client again produces the above behaviour only when the client used is an Android phone(running Android 10 in my case). However using a laptop to connect and disconnect the AP again and again does not affect the heap for some reason and hence no problem using that.
So the issue remains only when using an android phone to connect to the WLED-AP.

Adityakush, 2020-07-26:
I would also not mind giving up on some of the additional features in a custom compile if disabling something can free some of the heap.

Aircoookie, 2020-07-26:
Hi!
Thank you for the detailed report! It indeed seems like the access point code does not free some per-client heap after the client disconnected. 
You can disable some features with the defines in `wled. h`, for instance disabling Alexa will free about 2k of heap. But it is unlikely that it will solve the root problem, it will merely allow for one or two more clients before becoming unresponsive. Will look into it soon!

Adityakush, 2020-07-26:
Thanks @Aircoookie. Hope you solve it soon.

Adityakush, 2020-09-17:
Hi @Aircoookie. My kids love using WLED lights but the restart issue on disconnect/reconnect keeps muddling the experience.  Please fix this issue sooner. Really Appreciate your work . Thanks man.

closed 2021-06-16 by stale[bot]