wled/WLED #1061: WLED-AP never appears

jerkwagon, 2020-07-24:
Wled-ap never appears.

**To Reproduce**
Steps to reproduce the behavior, if consistently possible

**Expected behavior**
A clear and concise description of what you expected to happen.

**WLED version**
 - esp32s dev
 - Version 0.10.0,
 - Format Bianary

Steps, old board, which i used before for other arduino steps (esp32)  flashed WLED_0.10.0_ESP32

No wled-ap

flashed esp32_bootloader

no wled-ap

Flashed WLED_0.10.0_ESP32 and ERASE ALL DATA

no wled-ap

reflashed bootloader. 

no wled-ap

took out a brand new esp32s 

flashed bootloader

flashed WLED_0.10.0_ESP32

no wled-ap

[image]

in short, i have done the following, 

WIPE/flash bootloader
Flash wled

WIPE/Flash Bootloader
Wipe/Flash wled

Wipe/Flash Bootloader
flash Wled
Flash bootloader

flashbootloader
flash wled

still no wled-ap on my ssids
[image]

Thank you for your help!

jerkwagon, 2020-07-25:
I also tried on a brand new MacBook air, installed python, esptool.

Then ran esptool.py write_flash 0x0 ./esp32_bootloader.bin it flashed, but still no wled-ap

jerkwagon, 2020-07-25:
[image]

this is what the serial monitor sees after the reboot

raoulteeuwen, 2020-07-25:
Long shot, but can you try https://github.com/esphome/esphome-flasher/releases ? Prob not related, but found it through https://github.com/Aircoookie/WLED/issues/320 .

jerkwagon, 2020-07-25:
> Long shot, but can you try https://github.com/esphome/esphome-flasher/releases ? Prob not related, but found it through #320 .

Using 'COM3' as serial port.
Connecting........_____..
Detecting chip type... ESP32
Connecting.....

Chip Info:
 - Chip Family: ESP32
 - Chip Model: ESP32D0WDQ6 (revision 1)
 - Number of Cores: 2
 - Max CPU Frequency: 240MHz
 - Has Bluetooth: YES
 - Has Embedded Flash: NO
 - Has Factory-Calibrated ADC: YES
 - MAC Address: 24:62:AB:FE:1E:48
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
 - Flash Size: 4MB
Unexpected error: The firmware binary is invalid (magic byte=FF, should be E9)

didnt work,

raoulteeuwen, 2020-07-25:
Found https://github.com/esphome/esphome-core/issues/311 > could you please re-download the bin and try again?

jerkwagon, 2020-07-25:
Solved, I dont know anything about coding chips, i am decent and troubleshooting, turns out  the bootloader was being overwriten by the WLED?  I used different software on epressif site to get a downloader (its made for batch firmwares) so basically i told it to write the bootloader at 0x0 (which was the answer from the google search to issue #517 ) then i said to start writing the WLED .bin at 0x10000, it was just a long shot but it worked, i think i could of told it to write at a smaller nunber but honestly have no idea how that stuff works,  Maybe some techie person can say.   Anyway heres a shot of my procedure WLED HERE WE COME!~

[image]

esptool.py --port <port> --baud 115200 --after no_reset write_flash --flash_mode dio 0x10000  WLED_0.10.0_ESP32.bin just might do the job for you if have the same issue.

closed 2020-07-25 by jerkwagon

raoulteeuwen, 2020-07-25:
At https://github.com/Aircoookie/WLED/wiki/Install-WLED-binary it is stated the write on an ESP32 should start at what you ended up with...