wled/WLED #1062: Limit the times an animation runs (or stop cycling of animations at all)

OliverHi, 2020-07-24:
Right now all animations run endlessly again and again. I would like to be able to limit the amount of times an animation runs, or at least stop the cycling at all and only run it once.
I have seen several requests where this could be useful but the only way currently seems to be a custom animation. One example where this would be extremely useful is animating lights on and off with constant lights on in between.

Aircoookie, 2020-08-03:
I agree this would be useful to have.
Will be possible once we have a playlist feature!

Male20, 2021-01-18:
I am hoping to use WLED to control stairs light and use additional presets for holiday seasons(Halloween Christmas Etc).  What I really would like to do is to have an option to stop the animation once is finished . All the commands will be send via Home assistant based on proximity sensor triggers and preconditions.  What is progress of the playlist feature?

On top of all that I would like to mention that I am using WWA WS2812B  LED strip and one of Nice to have features would be a dedicated color wheel.

lamasek, 2021-05-10:
I suggest and want to finish playlist support to the WEB/APP (section like favorities). It will also solve limit animation/preset cycling runs.
For me best way is something like:
  1; API command, eg PL=1 to play preset 1; duration (in 1/10 s, allowing 0 will be nice - when you want more commands)
  2; -----------------||----------------------
  3; -----------------||----------------------
  ....

And also inputbox with "Repeat playlist: NUM" (0 = infinitely) will be nice.
Should be more playlists, but 1 is enough for me and now;-)

Ps. Thanks very much for all of your work.

ScumbagSteve, 2021-06-01:
Is there any plans to implement this in the near future?
I am looking for exactly this. I have set up two animations that are triggered via a motion sensor by iobroker. I want the lights to stay on after the last animation has fully lit them.

ScumbagSteve, 2021-06-02:
By coincidence I found out about the "playlist" feature, where you can limit the amount of cycles and define the time per preset.
This function needs to be sent via Json and there is no gui as far as I saw.

The only problem I have is that the transition time doesn't seem to work. Each preset is applied immediately. I think this topic here can be resolved but further documentation in the wiki about the transition would be great.

lamasek, 2021-06-02:
I saw it, but I don't want to have playlist over JSON - then I need 
another PC to manipulate it.

I want playlist in GUI and about week ago I suggested to Aircookie, if I 
fundraise money for him, if he will make it, but he did not answered yet.

I really need it. I am not able to make on my own as new feature of 
WLED, and without it is better for me write my desired effects from 
scratch in mycropython.

Joseph

Ion3, 2022-03-31:
This request would indeed be very useful!

blazoncek, 2022-03-31:
Now you have playlists. They solve the problem entirely.

closed 2022-03-31 by blazoncek

Ion3, 2022-03-31:
If I understand correct, playlists are only available using JSON:

https://kno.wled.ge/features/presets/

Does this really solve the problem entirely?

lamasek, 2022-03-31:
> I saw it, but I don't want to have playlist over JSON - then I need another PC to manipulate it. I want playlist in GUI and about week ago I suggested to Aircookie, if I fundraise money for him, if [...]

Playlist (without JSON) works for me at least 3 months. Thanks aircoookie.

blazoncek, 2022-03-31:
> If I understand correct, playlists are only available using JSON:
> Does this really solve the problem entirely?

You got that wrong and yes it does.

OliverHi, 2022-10-21:
Has anyone actually got this to work? My idea is still to "wipe" LEDs on one by one and then keep all of them on as a "on animation". I am also looking for the same thing in reverse as an "off animation".
I tried to create the on animation by creating two presets: one using the wipe animation and one has all lights on. Then I created a playlist running wipe on one time and then the just on one as end preset.
Unfortunately this is not working at all as intended because the wipe animation starts at a different point every time and I also have to trial and error the time it takes to run it exactly once to on. I also have no clue at all how to get the off animation. Would be glad for any feedback on how to do this. Maybe @blazoncek can help?

blazoncek, 2022-10-21:
Add a *timebase* to your preset API.
Look into [KB](https://kno.wled.ge/interfaces/json-api/).

OliverHi, 2022-10-21:
Thank you, that helped a lot. For those looking into the same problem: you have to create a new preset, then check api command field and add `"tb":0,` to the existing command to do that.
Do you happen to know how the reverse (starting at all LEDs on, then wiping effect, then all off) would be possible? I tried using the same preset but with `"rev": true` but that just changes the direction of the on effect, not sure how to make it turn off instead. I tried playing around with the timebase here too but that does not seem to help.

blazoncek, 2022-10-21:
You can swap colors 1 & 2 and reverse the segment

kapellmeister84, 2022-11-28:
> Thank you, that helped a lot. For those looking into the same problem: you have to create a new preset, then check api command field and add `"tb":0,` to the existing command to do that.
> Do you happen to know how the reverse (starting at all LEDs on, then wiping effect off one by one, then all off) would be possible? I tried using the same preset but with `"rev": true` but that just changes the direction of the on effect, not sure how to make it turn off instead. I tried playing around with the timebase here too but that does not seem to help.

Hey man, im just looking into the same problem. I tried adding the „tb“:0 but I don’t seem to find the right spot to insert it.. I just won’t work… can you post an example or explain a little bit more, for stupids like me? Thanks a lot!

blazoncek, 2022-11-29:
@kapellmeister84 please use WLED [form](https://wled.discourse.group) or Discord for help and support questions.