wled/WLED #1066: WS2801 Working but gives wled error2

insproad, 2020-07-25:
Hi, thats awesome project but when I  use WLED_0.10.0_ESP8266_ws2801.bin app always gives wled error 2 and when I power off wifi forgettting from esp8266. I need to write wifi information every reset time.

raoulteeuwen, 2020-07-25:
Does https://github.com/Aircoookie/WLED/issues/1014 help?

insproad, 2020-07-26:
I also use ESP Home Flasher and I cant use different bin for WS2801. Other bin files dont work with ws2801.

Aircoookie, 2020-07-26:
What is your board? The WS2801 binary is compiled for 4M boards like wemos d1 mini and the NodeMCU

insproad, 2020-07-26:
My esp model is esp-01 black.

insproad, 2020-07-30:
Have you any suggestion?

Aircoookie, 2020-08-03:
Hi, esp-01 only has 1mb of flash memory, so the default binary won't work.
I compiled a custom one for you:
https://drive.google.com/file/d/1_AzpjjOzXPZyEXoQzxX8QkZF-hsnWDMB/view?usp=sharing
Hope it works well :)

insproad, 2020-08-03:
> Hi, esp-01 only has 1mb of flash memory, so the default binary won't work.
> I compiled a custom one for you:
> https://drive.google.com/file/d/1_AzpjjOzXPZyEXoQzxX8QkZF-hsnWDMB/view?usp=sharing
> Hope it works well :)

You are awesome my friend I friend and now doesnt have error2 and forget wireless. Working perfect:))))
Can you help me for google home connection too??

WoodyLetsCode, 2020-08-05:
Hey,
join the WLED Discord server. There is a guide for google home. 😄

RajeeRah, 2020-08-31:
Hi Guys, where can i find the wiring to WS2801 ? i tried using this link which doesnt work: https://alexbloggt.com/wp-content/uploads/2015/10/nodemcu_pinout_700.png

I did connected power via USB
LED GND -> GND
LED clock -> SCLK
LED Data -> MOSI

RajeeRah, 2020-08-31:
> Hi Guys, where can i find the wiring to WS2801 ? i tried using this link which doesnt work: https://alexbloggt.com/wp-content/uploads/2015/10/nodemcu_pinout_700.png
> 
> I did connected power via USB
> LED GND -> GND
> LED clock -> SCLK
> LED Data -> MOSI
> 
> tried also:
> LED clock -> D3
> LED DATA -> D4

Finally works - i had to switch clock and DATA + powered the nodemcu also via 5V from LED strip. Thanks for the amazing project. :)

newwolf, 2020-09-30:
Experienced the same issue with my Wemos D1 mini and WS2801.
Flashing "WLED_0.10.2_ESP8266_ws2801.bin" with Tasmotizer gave me Error 2 and loosing Wifi Info.
But flashing "WLED_0.10.2_ESP8266_ws2801.bin" with "ESP8266Flasher" on 0x00000 works perfect and without errors.

closed 2021-02-06 by stale[bot]