wled/WLED #1071: Won’t show in my network

developinfro, 2020-07-28:
As I'm new to all of this so please bare with me. I have flashed a D1 mini V2 with v0.10.0 wLed. Hotspot shows up I can login and control leds. When I set my network save and reboot it never shows up on my router. I've verified that my network is 2.4 GHz. And have flashed node with Tasmota and it worked. So I assume the chip is working. I have read everything I can find the internet about this with no luck. My hope is that someone here can help. My plan is to use WLed with Hyperion.

Thank in advance

raoulteeuwen, 2020-07-28:
- Are you sure the name and pwd for your wifi gets inserted as you want, so no first character auto-capitalized while you type it? No spaces leading or trailing?
- The signal strength could not be a problem? 
- Can you confirm/try using channel 1 for the wifi signal? 
- You're not doing any filtering on mac address on the AP/router? 
- What brand and model router do you have?

toto79, 2020-07-28:
@developinfro : see #1024 and #1030 , maybe you run into the same error.
Also you should check that your are using the right binary

developinfro, 2020-07-29:
> - Are you sure the name and pwd for your wifi gets inserted as you want, so no first character auto-capitalized while you type it? No spaces leading or trailing?
>
> - The signal strength could not be [...]

I'm sure name and passwd are correct
Signal strength shouldn't be an issue. I have google WiFi nested with 4 access points and I'm not but a few feet from one of the routers.
I'm not sure what you mean by channel 1 or filtering on MAC address AP/router

raoulteeuwen, 2020-07-29:
If you don't know what MAC filtering is, you probably are not doing it ;-) (every device has a MAC address: for security you can limit access to your AP/wifi/LAN to known MAC addresses).

Re channel 1: if you sign in to your router, you should be able to specifiy at what channel your Wifi signal is broadcasted: it would be nice to check if you get a working WLED-nodeMCU when the AP is broadcasting the 2.4Ghz on Channel 1.

developinfro, 2020-08-03:
I wanted to say thanks for all your help with this. My problem was Google WiFi. Change to my spectrum Router and was able to get IP address. Thanks again this issue can be closed.

Aircoookie, 2020-08-03:
Interesting that the router was the culprit. I wonder if it is a configuration problem or a problem with Google WiFi itself. Would be interesting to know if someone is running WLED successfully with that mesh system. Maybe it is fixable, since I would like to achieve maximum compatibility :)

CMaxim77, 2020-08-03:
I run WLED with google WiFi and for the most part have no issues with controlling the lights. Every once in a while I have to unplug the esp board then everything comes back just fine.

developinfro, 2020-08-04:
CMaxium77

Are you running your Wi-Fi with a mesh?

CMaxim77, 2020-08-04:
I currently only have a single Google Hub, but I'm planning on adding 3 more tonight. If I run into any connection issues regarding WLED I'll be sure to let you know.

forresthopkinsa, 2020-08-10:
@CMaxim77 Any updates?

CMaxim77, 2020-08-10:
I'm sorry, it took me so long to reply. After adding 3 more google WiFi routers to my network I can confirm that wled is working so far in my mesh setup. If I run into any issues I'll be sure to post.

closed 2020-12-16 by stale[bot]