wled/WLED #1072: Effect: Round-LED-Clock (feature request)

kolbenhans, 2020-07-28:
**Is your feature request related to a problem? Please describe.**
No, not ab problem. Only a possible nice effect to configure

**Describe the solution you'd like**
a round LED clock lightning based on actual ntp time.

**Describe alternatives you've considered**
see the following project: https://github.com/leonvandenbeukel/Round-LED-Clock

Aircoookie, 2020-07-28:
Hi!
This feature is already part of WLED - just select Analog Clock in the drop down menu in Time Settings! Don't forget to also turn on NTP :)

sansillusion, 2020-07-28:
Yes I made one for a friend was great ! Used 60 leds.

Just posted it to youtube :
https://youtu.be/It_TK4mL4nk

TGM3D, 2020-08-04:
or look in this graetful project whit a clock and the WLED Software !

closed 2020-08-04 by Aircoookie