wled/WLED #1073: Current calculation

Appie-rappie, 2020-07-30:
WLED version 0.10.0

Current idle current estimation is 100mA + 1 mA per LED (if I am correct).

For those of us who use a relay, there is no idle current per LED.

_Suggestion_:
- Add 'off by relay' switch to LED settings
- Calculate estimated idle current as 80 mA (more realistic idle current for ESP8266) + 1 mA per LED **only** if 'off by relay' is set to false
- No changes needed to estimated LED current when working

huggy-d1, 2020-07-30:
Do you know what wled file and which line of code does this calculation incorrectly?

fishbone-git, 2020-07-30:
fx_fcn.cpp ~line204.is where the ESP constant is set. The rest follows below to calculate based on brightness and type

Appie-rappie, 2020-07-30:
That's correct, file is FX_fcn.cpp. I think the actual line no for the ESP constant is 202 (in version 0.10.0).

And line 273 adds one mA per LED to standby power usage, which is correct without use of relay, but can be omitted if relay is used.

huggy-d1, 2020-07-31:
At the very minimum, the power consumption calculation could check if the Relay Pin is at least defined and ≥ 0.  Something like:
```
#if RLYPIN <= 0
currentMilliamps += _length;
#endif
```

fishbone-git, 2020-07-31:
except that the default code has the relay built in, but cannot tell if a user is making use of that relay.  Might be better to hold off the power calculation until the selectable pins feature is available.  At least then, it should be reasonable to have the relay off by default and make the reasonable assumption that if it is selected with a pin that it is in use and adjust the current estimate accordingly,

huggy-d1, 2020-07-31:
Agreed, need a user setting to note if an external relay is used before adjusting the power consumption calculations.

huggy-d1, 2020-08-03:
As an alternative, you can add 1ma x # LEDs to the current limit you input. This works today without any code changes.

Appie-rappie, 2020-08-03:
Would that adjust the power consumption to solely the ESP current in idle mode if you use a relay???!

closed 2020-12-09 by stale[bot]