wled/WLED #1074: ESP32 + SK6812 RGBW starts flashing half way 480 leds

MvZeeland, 2020-08-01:
Dear,

Just finished my stairwells project with an ESP 32 with 480 SK6812 RGBW (60leds/meter) and an 30 Amp/5v power supply.

However, it seems that arround half way through the leds, it starts to blink/flash. Could this be the result of not using any levelshifters/translators? i've already tried to play with the Automatic Brightness Limiter but have no good result. Also the blinking/flashing effect occurs mainly when using an solid color, especially when using an white color.

Please see this video; 
[https://youtu.be/dc8RYA9O6sU](url)

My schematics are as following;
[image]

Thanks in advance!

raoulteeuwen, 2020-08-01:
Afaik, the data signal should be repeated and 'rejuveniled' by each led. Are you sure the data line is connected correctly around the position where the flicker starts?

MvZeeland, 2020-08-01:
Hi Raoul,

Thanks for your reply.

I've measured every data to data connection with an multimeter and all diode values are roughly the same, which should indicate that there is not an loose cable.

However, i've flipped the cables from data in and data out so the first led is at the bottom. This gave me roughly the same results.

I've been using these connectors in every stairs for connecting the wires.
[image]

I've checked every strip from top to down, everyone seems to have an solid connection. However it's very brittle.. should i need to hardly solder the connections? That'll take some time and effort..

Thanks in advance for any help!

sansillusion, 2020-08-01:
From your picture, I think you may need power injection at both ends if the string is too long.

KensonN, 2020-08-02:
Maybe try connecting the ground of the led strips straight to the ESP ground. Was having similar issues

MvZeeland, 2020-08-02:
> Maybe try connecting the ground of the led strips straight to the ESP ground. Was having similar issues

Thanks for your reply, ESP and led strip share the same ground. Could you explain it in some more detail? Thanks!

KensonN, 2020-08-02:
Yeah, try adding an additional ground wire from the LED strip straight to the ESP

MvZeeland, 2020-08-02:
That did the trick! However when i connected only an ground wire, it did some flashing during the color/effect changing. When i added the positive wire also, it was fixed!

Thanks KensonN, sansillusion & raoulteeuwen for your support!

closed 2020-08-02 by MvZeeland

KensonN, 2020-08-02:
Glad that worked!