wled/WLED #1075: Set Current Preset Cycle As Boot Default Issue

Model-Man, 2020-08-02:
**WLED version** 0.10.0  Build 2005030
 - Board:  Wemos D1, ESP32 dev]mini clone

Hi all. First up a big thank you to Aircoookie for this great software and app. Also to all the contributors who have helped him develop what I believe is one of the best led pixel controllers out there.

A quick question regarding the "Set current preset cycle as boot default".  While checking the box and saving this option the box remains un-checked so the preset doesn't run. I've seen this issue mentioned before on earlier builds however thought it was fixed on the current version.
Is this a current app bug or only a problem on certain android devices?

Thank you for your help!

Aircoookie, 2020-08-03:
Hi, you’re welcome, I’m very happy that you like WLED!

Did you have the preset cycle active while saving the LED settings?
Perhaps counterintuitively, this saves the state as-is (if preset cycle is on and it is checked it will run on next boot, if the cycle is off and it is checked it will not run). Therefore, it will also automatically un-check every time the settings are saved.

Thank you for the feedback though, I believe a `checked = preset cycle runs on boot`, `unchecked = it doesn't run` would be more logical.

Model-Man, 2020-08-03:
Hi and thanks for your rapid response. I may of had some presets running at the time of attempting to save a preset cycle boot up. I have yet to keep the "Set current preset cycle as boot default" box checked as it always disappears after a save. 
I did discoverer on one of my test controllers that the boot preset cycle did in fact run OK on a cold boot however on the app the "tick" box still remained unchecked.  I'll do some further testing as this is only my third day of using WLED. I except that doing things in a different order may confuse certain settings.  I have a test bench with multiple arrays of addressable pixels and controllers so I'll report back on what I find.  

Many thanks again.

Aircoookie, 2020-08-03:
No worries!
As I said, the checkbox will always uncheck automatically. WLED will use the way the preset cycle is currently set up as boot default if you check the checkbox. If you then turn off Preset Cycle, it will still turn on at every boot. You will need to check the box again with Preset cycle turned off to also disable it at boot :)

Model-Man, 2020-08-04:
Ok I did some testing using your method and got mixed results.

I made 6 presets then ran 1-3 and stored using your instructions.  On boot up it ran presets 1-3 fine.
Next test was to change the presets from 4-6 and store.
On boot up it ran presets 1-3 then 4-6 and then continuously ran 4-6 as expected due to it being the last stored preset. Why did it run the first stored preset 1-3 as well during boot up?

And finally the "Transition Time" in Seconds keeps defaulting back to 0.7s no matter how many times I try to change it. The lights respond initially to the adjusted time of 1.5s however revert straight back to 0.7s on returning to the presets page within the app.
The modified "Time Per Preset" and "Preset Cycle" "First" and "Last" store perfectly.
I make the changes and then "check" the preset cycle box last.  Any idea why only the transition time won't store?

Thanks in advance and sorry for all the questions. I'm sure there's a simple answer... 👍

Model-Man, 2020-08-10:
After much testing I think everything is kinda sorted and I hope this helps others that might encounter the same issue. 
With regards to the following and please correct me if I'm wrong.

Presets

1) Presets that are currently stored on a cold boot must be un-checked and re-saved to erase the memory. Only then can new presents be stored by again re-checking the box. Not doing it this way seemed to mingle old and new presets together.
It's a two stage process of erasing old then re-adding new each time you want different presents. 

Transition Time

2) You have effectively two transition features available. The first on the preset page only adjusts the transition time for the current session. It is not permanently stored and will revert straight back to 0.7 sec when you exit the app page.  (This was my  mistake)
To permanently store a Transition/ crossfade you have to enter the time in the "led settings page" , setting it in ms. In my case 1500ms or 1.5 sec.

Sorted however just needed some time learning the little quirks and different ways of doing things compared to other controllers I've tested.. 👍

Dilftastic, 2020-10-06:
I would like to add to this and get an answer. Although start up on boot is nice. I have set presets for 1 thru 4. I make a preset cycle to cycle through 1 thru 4. And I have checked for it to start at boot up. However, i have constant power going to my node mcu. And have macros to simply turn on and off my lights. Unfortunately, it seems T=1 and 0 turn them off and on but wont start my preset cycle. Is there a macro I can write that will say turn on lights and start preset cycle?

ssshake, 2020-11-19:
I am having the opposite issue. I turned on preset cycles at boot, saved it and that works fine. Now I want to shut it off and I can't. I've unchecked the preset cycles in the fav's menu and turned it off in LED preferences menu and hit save. But every boot preset cycles is still on and running

corkymork, 2020-12-30:
Having a similar, but different issue.
After setting up a preset cycle, and letting it run, I check "Set current preset cycle as boot default". After a rest or power cycle, my LEDs are off. If I nudge the brightness slider on the "colors" page, the LED go crazy. Turns out that the time per preset gets set to zero, so it zooms through all the presets! Other preset cycle values remain unchanged.
Any thoughts?

closed 2021-06-16 by stale[bot]