wled/WLED #1077: Multiple WLED clients and animations

Justdigit, 2020-08-04:
First of all thanks for this great RGB software.
I run this on my PC to control casefans. It's way better than all of those property software of all those manufacturers. 

I have hooked up all fans with digital in and digital out. So I can get nice animations. 

But I have two devices the graphic card and a strip with no digital out. 
One is on the same esp8266 as end device. And the other one is on a separate esp8266 with sync.

Unfortunately and of course it won't sync animations like bouncing balls. On the second esp8266.

Would it be possible to mod the software to sync the animations on the second esp8266 when you select an animation on the main esp8266?

Aircoookie, 2020-08-04:
Hi, awesome that you find WLED useful! 
Yeah, while the UDP sync will sync colors and the effects mode, some effects based on randomness will be unique to each controller. It is not easily possible to sync them because even though we could share a random seed between the controllers, we would need to make sure that the number of calls to random() is the same on each controller. What I would recommend is that you only use a single controller and run the output in parallel so that both strips display exactly the same :)

Justdigit, 2020-08-04:
> Hi, awesome that you find WLED useful!
> Yeah, while the UDP sync will sync colors and the effects mode, some effects based on randomness will be unique to each controller. It is not easily possible [...]

Thanks.... I will do that maybe.... 

I saw another thread with esp32 with more than one channel out. I saw its still in development.

Is that implementation going to sync animations? Or is it the same as a segment.

closed 2020-08-04 by Justdigit