wled/WLED #1079: Alexa in Networks whith more one WLED installation Problem

TGM3D, 2020-08-04:
At the moment I have 4 LED chains and lamps etc. with 4 WLED installations in my own network. All devices have their own Alexa names and are set up separately. Nevertheless, everyone always turns on or off with commands. I have switched off the sync function. What else can you set or is it currently a malfunction?
On this occasion, thanks again for the extremely flexible and great software!

huggy-d1, 2020-08-04:
Change the Sync UDP ports so they are different numbers on each device.  Then no matter what is going on with the sync functions, the sync messages will be ignored.  If that works, please post back - and it means they are still using WLED Sync feature.

In your she-who-shall-not-be-named configuration for WLED devices, are they in a group (room, floor, or scene)?  Or are they 4 unique stand-alone "_she_" devices?  Sometimes "_she_" gets too helpful when parsing what we say.  For example, tell she-who-shall-not-be-named to turn on the lights and every "_she_" enabled device with the word "light" in its name gets an ON command.  I believe this is the way it works in general.

It can also be a grouping, like all WLED devices are upstairs, and you say "turn on the upstairs LED strip in the closet" and "_she_" gives up after "upstairs LED" and turns them all on.

These are things to consider when naming, grouping, and even when interacting with she-who-shall-not-be-named.

TGM3D, 2020-08-04:
Thanks for the quick help.
It is sufficient to deactivate the function: Send Alexa notifications :. Then you can still activate or deactivate the sync function as required. Changing the UDP ports also leads to success, but removes this sync function. Incidentally, whether the Alexa device was in certain groups was irrelevant.
Thanks

closed 2020-08-04 by TGM3D

huggy-d1, 2020-08-04:
Thanks!  I knew I was stretching it way far with the "_she_" groupings and namings, but I've had "_she_" do some odd things when trying to answer someone in the room while trying to request a device action, so tossed that in there for good measure.