wled/WLED #1080: ESP8266, multiple segments have to be saved with slot16 to work after a reboot

fribse, 2020-08-05:
**Describe the bug**
I noticed recently that my segments were dissappearing on the only ESP8266 I have running WLED 0.100.0, I tried different things which didn't help,  I then tried setting the default behaviour for the two segments I have defined in slot 16, and then it keeps the segments.

**To Reproduce**
Bare ESP8266 with WLED, two segments set up, reboot, they are gone.

**Expected behavior**
A reboot shouldn't remove the segments

**WLED version**
 - Board: D1 mini (esp8266)
 - Version 0.10.0
 - Format binary

huggy-d1, 2020-08-05:
I believe the behavior you experience is known. I think it may even be documented as to how it currently works - because of memory constraints in the current software design configuration.
In a couple of versions into the future, I believe this behavior will not happen because things will start to feel more predictable as opposed to how they feel now - like they are dropping segment setups.

garyd9, 2020-08-09:
From https://github.com/Aircoookie/WLED/wiki/Presets

There are 16 preset slots in total. In 0.9.0, **the last preset (16) is capable of saving the entire segment configuration. All other presets only save a single segment** (main segment, the first one by default) and restore that preset to all selected segments.

(emphasis added)

closed 2020-12-08 by fribse