wled/WLED #1081: Philips hue error when connecting to bridge

evolutionaryman, 2020-08-05:
An error occurs on trying to link to the hue bridge.

Hue status: Invalid light id.

I attempted this 5 times with the same results. Anyone else having this problem? Checked my IP address and tried to fix many different ways.

Aircoookie, 2020-08-06:
Hi!
What have you entered in the light ID field? It needs to contain the ID of the Hue light you want WLED to sync to.
You can find it by going to the light list in the About section of the Hue app. The number in front of the light is the one you'll need to enter into WLED settings :)

huggy-d1, 2020-08-06:
Maybe @evolutionaryman thinks that WLED gets added to a Philips Hue bridge as another Hue bulb?
That is not what WLED does (at least I do not think so..).
My understanding is WLED emulates a Hue light so Alexa can control it (without the cloud).

evolutionaryman, 2020-08-06:
Your right. I miss understood how the hue sync worked. All cleared up now. Thank you so much!

closed 2020-08-06 by evolutionaryman