wled/WLED #1083: LEDs start to glow after a while

t6a, 2020-08-06:
Hello!
after a day or so, the first LED starts to glow. Often green, sometimes red. After a couple of days (non-usage) the whole strip glows green - but the app shows that the Stripe is off. When I turn on the stripe and back off, it stays off for a while and then it begins again to glow.

Hardware:
- Wemos D1 Pro
- WS2815 

[image]

huggy-d1, 2020-08-06:
If I were to guess, it is a wiring issue.
Make sure the common (0V or GND) wire on the controller is electrically connected to the power supply 0V going to the LED strip.
If you have a multimeter, measure the voltage from the ground on the controller power to the led strip common.  If you see the LED suddenly act normal, my suggestion will fix the problem.

t6a, 2020-08-07:
Thanks - it was a wiring issue.
I implemented a level shifter and now it’s fine.

Do I still need a capacitor or resistor (with WS2815, 12V)?

closed 2020-08-07 by t6a