wled/WLED #1084: Alternative color picker: Enter RGB(W) decimal or hex code

gman-php, 2020-08-06:
**Is your feature request related to a problem? Please describe.**
A perfectionism problem I would say: I use WLED to provide ambient light to my shelf holding my console collection. Every console gets a background light with their companies main color, And e,g, SEGA and PlayStation have a very similar blue. So it is really hard to just pick them with the color picker.

**Describe the solution you'd like**
Instead of just a color picker, it could be an option to switch between the picker and manual RGB(W) code text fields. LIke all image applications for example do.

**Describe alternatives you've considered**
A way to edit the config behind manually maybe?

**Additional context**
N/A

Thank you for your ideas for making WLED better!

huggy-d1, 2020-08-07:
Not sure manually is what I would describe it, but you can create URLs on another device to select very specific color values.  Perhaps you could iterate over a few URLs until you find the exact colors you want, then store them in presets.
Basically you would be manually color mapping requested color to actual visual color output which relates to room lighting, time of day (if outside light can enter the room), your visual acuity (which is perfectly suited to your needs !), the strip's performance (delivering proper current), and the LEDs used on the strip.
Look at the WLED Wiki for URL requests that can help you find the right colors.

gman-php, 2020-08-09:
Yah, JSON API is a good idea. I can use that to exactly set the color codes, thx! :) Still, I think it would be really convenient to "just" have three/four text inputs in the UI!

The JSON API now will fix it for me though, thx!

JeffResc, 2020-08-10:
Hey @gman-php, I just created pull request #1093. If you have a minute, let me know what you think of the addition.

gman-php, 2020-08-10:
Hey @JeffResc,

that looks awesome, thanks for the fast solution!
I didn't comment on the PR itself, as I wanted to leave that for code/code style/potential bug comments. And comment here if that solves my feature request :)

Pretty cool, thx!
Greetings,

Andy!

blazoncek, 2022-07-25:
Implemented way back in 2020. Closing.

closed 2022-07-25 by blazoncek