wled/WLED #1086: bei aktiviertem MQTT ständiges an/abmelden des WLAN an Router

eldoman, 2020-08-08:
Hallo, 
sobald ich MQTT aktiviere meldet sich das WLAN des NodeMCU alle paar Sekunden an dem Router an und ab.
Die Verbindung zum MQTT Server (IOBroker) steht.
Sobald ich MQTT deaktiviere (durch ESPHome-Flasher-1.2.0-windows-x64.exe) funktioniert das WLAN des NodeMCU wieder zuverlässig. Dies habe ich jetzt mehrmals ausprobiert.

 - Board:    NodeMcu ESP8266 ESP
 - Version:  WLED_0.10.0_ESP8266.bin
 - Format:  Binary

Trotzdem ein sehr cooles Programm.
Viele Grüße und weiter so! :-)

StefanRu1, 2020-08-10:
Hi eldoman,
das ist sehr seltsam. Ich nutze hier eine Fritzbox, den Mosquito MQTT Server auf einem Rapberry und WLED auf einem WEMOS D1 mini.
Ich habe das Problem nicht.
Jetzt wäre es interessant was bei dir genau passiert.
Kannst du eventuell deine NodeMcu mit einem USB Kabel am PC anschließen und den serial Monitor vom Arduino Studio mit anmachen? Dann sollte man sehen was da passiert.

Gruß,
Stefan

droloff, 2020-08-17:
Ich hab hier einen ganz ähnlichen Efekt.
Wenn ich die Gegenstelle (Iobrocker) beende ist alles gut, sobald eine MQTT Verbindung besteht bekomme ich folgenden output:
Leider kann ich den Stack nicht interpretieren. Gibts da ein Tool?

```
---WLED 0.10.1 2007190 INIT---
esp8266 2_7_1
heap 34248
LEDs inited. heap usage ~3000
Load EEPROM
Ada
Access point disabled.
Connecting to PrivateSpace...

Connected! IP address: 192.168.178.122
Init STA interfaces
mDNS started
Reconnecting MQTT
MQTT ready
Publish MQTT
MQTT msg: wled/test

Exception (28):
epc1=0x4000e1e0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffeb30 end: 3fffffb0 offset: 0190
3fffecc0:  3fff1b90 3fff5f0c 3fff1b90 40233530
3fffecd0:  3ffef938 3fff5f0c 00000000 40215dc7
3fffece0:  00000008 00000000 01ac0f00 02ac0f00
3fffecf0:  03ac0f00 04ac0f00 3fff5a8c 3fffee80
3fffed00:  00000000 3fff2cbc 00000000 40237e2a
3fffed10:  3fff6086 40105797 3fff604c 3fff604c
3fffed20:  00000002 00000000 00000020 4021a824
3fffed30:  00000000 40263894 00000002 40101e94
3fffed40:  40215d4c 4010512f 40237e34 40237e14
3fffed50:  00000000 4010430a 3ffef568 40263b20
3fffed60:  3fff2cbc 3fff5f0c 00000000 40100e20
3fffed70:  00000000 4bc6a7f0 a5e353f7 00000000
3fffed80:  3ffedc22 00000079 3fff58ac 40237fb5
3fffed90:  00000000 00000000 00000000 00000000
3fffeda0:  3ffedc22 4021a433 3fffedf0 4021b75d
3fffedb0:  00000000 00000000 00000000 00000000
3fffedc0:  000038d0 00000079 3fff58ac 4021b805
3fffedd0:  3ffedc22 00000079 3fff2adc 4021a591
3fffede0:  3fff560c 4bc6a7f0 40219908 40237ef8
3fffedf0:  3fff2d0c 00000000 402198bc 40237f7c
3fffee00:  0000000d 4bc6a7f0 40100744 00001d9a
3fffee10:  3fff2c5d 3fff58ac 4bc6a7f0 00000000
3fffee20:  00223aca 00000000 40100744 00000000
3fffee30:  3fff2adc 3fff5624 00000000 4021a5c4
3fffee40:  3fff2adc 3fff5624 3fff258c 40226f64
3fffee50:  3fff5f94 0023c695 3fff5f94 3fff2588
3fffee60:  3fff5f94 0000006d 3fff2adc 3fff2588
3fffee70:  3fff5f94 3fff2590 3fff258c 40226fa8
3fffee80:  3fff5f94 3fff2590 3fff258c 40266398
3fffee90:  00000079 7ab2a8c0 00000018 00000079
3fffeea0:  00000000 00000018 3ffeee30 401009d0
3fffeeb0:  00000000 00000000 3ffec38d 3fff2768
3fffeec0:  3ffedbfa 3fff2770 3fff5624 40264901
3fffeed0:  3fff24ac 3fff5e54 3fff5e54 4010064c
3fffeee0:  00000008 3fff5a8c 3ffeee30 3fff5e54
3fffeef0:  3ffedbec 00000000 3fff5624 40263e41
3fffef00:  57b2a8c0 000000d7 00000000 0000001e
3fffef10:  00000002 0000001a 4025467b 3ffeee30
3fffef20:  3ffedbc4 3fffdcc0 3ffeb4b0 3ffeb4b0
3fffef30:  402545ee 3ffeee30 00000000 3fff5e9c
3fffef40:  3fffdc80 00000000 3fff5624 40259fcf
[... 260 lines cut ...]
3fffff90:  3fffdad0 00000000 3fff1c9c 4021f158
3fffffa0:  feefeffe feefeffe feefeffe 402358f0
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3456, room 16
tail 0
chksum 0x84
csum 0x84
va5432625
~ld

---WLED 0.10.1 2007190 INIT---
esp8266 2_7_1
heap 34248
LEDs inited. heap usage ~3000
Load EEPROM
Ada
Access point disabled.

```

viele Grüße
Dirk

Aircoookie, 2020-08-18:
@droloff Yep, es gibt da das Tool EspExceptionDecoder, funktioniert nur wenn man selbst kompiliert hat.
Exception 28 ist tatsächlich häufiger, oft hilft es, in den Sync Einstellungen den Alexa-Support zu deaktivieren...
Hoffe du bekommst es ans Laufen! Sehe aber auch gerade dass offenbar ca. 1000 LEDs konfiguriert werden, das kann auf dem 8266 in Verbindung mit MQTT sicherlich zu RAM-Problemen führen. In dem Fall würde ich einen ESP32 empfehlen.
Viele Grüße!

droloff, 2020-08-19:
Ich hab mal einen Stacktrace decodoiert und eine PR #1101 eingestellt.

```Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
Decoding 114 results
0x40233530: Print::println(char const*) at C:\Users\Dirk\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Print.cpp line 198
0x40215dc7: onMqttMessage(char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int) at D:\Daten\Dirk\Documents\Arduino\WLED/wled00/mqtt.cpp line 73
0x402541a3: ppTxPkt at ?? line ?
0x4024772f: ieee80211_output_pbuf at ?? line ?
0x40237e2a: std::_Function_handler ::_M_invoke(std::_Any_data const&, char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int) at c:\users\dirk\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2/functional line 2073
0x40105797: wdt_feed at ?? line ?
0x4021a824: AsyncMqttClient::_onMessage(char*, char*, unsigned char, bool, bool, unsigned int, unsigned int, unsigned int, unsigned short) at c:\users\dirk\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2/functional line 2174
:  (inlined by) AsyncMqttClient::_onMessage(char*, char*, unsigned char, bool, bool, unsigned int, unsigned int, unsigned int, unsigned short) at C:\Users\Dirk\Documents\Arduino\WLED-master/wled00\src\dependencies\async-mqtt-client/AsyncMqttClient.cpp line 523
0x40263894: etharp_send_ip at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip/src/netif/etharp.c line 435
0x40215d4c: onMqttMessage(char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int) at D:\Daten\Dirk\Documents\Arduino\WLED/wled00/mqtt.cpp line 55
0x40237e34: std::_Function_base::_Base_manager ::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\dirk\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2/functional line 1934
0x40237e14: std::_Function_handler ::_M_invoke(std::_Any_data const&, char*, char*, AsyncMqttClientMessageProperties, unsigned int, unsigned int, unsigned int) at c:\users\dirk\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2/functional line 2069
0x40263b20: etharp_output_to_arp_index at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip/src/netif/etharp.c line 890
0x40100e20: umm_free_core at C:\Users\Dirk\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc/umm_malloc.cpp line 316
0x40237fb5: std::_Function_handler    (AsyncMqttClient*, std::_Placeholder1>, std::_Placeholder2>, std::_Placeholder3>, std::_Placeholder4>, std::_Placeholder5>, std::_Placeholder6>, std::_Placeholder7>, std::_Placeholder8>, std::_Placeholder9>)> >::_M_invoke(std::_Any_data const&, char*, char*, unsigned char, bool, bool, unsigned int, unsigned int, unsigned int, unsigned short) at c:\users\dirk\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2/functional line 2073
0x4021a433: AsyncMqttClient::_onData(AsyncClient*, char*, unsigned int) at C:\Users\Dirk\Documents\Arduino\WLED-master/wled00\src\dependencies\async-mqtt-client/AsyncMqttClient.cpp line 402
0x4021b75d: AsyncMqttClientInternals::PublishPacket::_preparePayloadHandling(unsigned int) at C:\Users\Dirk\Documents\Arduino\WLED-master/wled00\src\dependencies\async-mqtt-client\AsyncMqttClient\Packets/PublishPacket.cpp line 72
0x4021b805: AsyncMqttClientInternals::PublishPacket::parseVariableHeader(char*, unsigned int, unsigned int*) at C:\Users\Dirk\Documents\Arduino\WLED-master/wled00\src\dependencies\async-mqtt-client\AsyncMqttClient\Packets/PublishPacket.cpp line 55
0x4021a591: AsyncMqttClient::_onData(AsyncClient*, char*, unsigned int) at C:\Users\Dirk\Documents\Arduino\WLED-master/wled00\src\dependencies\async-mqtt-client/AsyncMqttClient.cpp line 440
0x40219908: std::_Function_base::_Base_manager    (AsyncMqttClient*, std::_Placeholder1>, std::_Placeholder2>)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\dirk\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2/functional line 1931
0x40237ef8: std::_Function_handler    (AsyncMqttClient*, std::_Placeholder1>, std::_Placeholder2>)> >::_M_invoke(std::_Any_data const&, unsigned short, unsigned char) at c:\users\dirk\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2/functional line 1893
:  (inlined by) std::_Function_handler    (AsyncMqttClient*, std::_Placeholder1>, std::_Placeholder2>)> >::_M_invoke(std::_Any_data const&, unsigned short, unsigned char) at c:\users\dirk\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2/functional line 2071
0x402198bc: std::_Function_base::_Base_manager    (AsyncMqttClient*, std::_Placeholder1>, std::_Placeholder2>, std::_Placeholder3>, std::_Placeholder4>, std::_Placeholder5>, std::_Placeholder6>, std::_Placeholder7>, std::_Placeholder8>, std::_Placeholder9>)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\dirk\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\4.8.2/functional line 1931
0x40237f7c: std::_Function_handler    (AsyncMqttClient*, std::_Placeholder1>, std::_Placeholder2>, std::_Placeholder3>, std::_Placeholder4>, std::_Placeholder5>, std::_Placeholder6>, std::_Placeholder7>, std::_Placeholder8>, std::_Placeholder9>)> >::_M_invoke(std::_Any_data const&, char*, char*, unsigned char, bool, bool, unsigned int, unsigned int, unsign [... cut ...]
:  (inlined by) esp_yield_within_cont at C:\Users\Dirk\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp line 111
:  (inlined by) __yield at C:\Users\Dirk\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp line 130
0x4021e685: WLED::loop() at D:\Daten\Dirk\Documents\Arduino\WLED/wled00/wled.cpp line 60
0x4021f158: loop at D:/Daten/Dirk/Documents/Arduino/WLED/wled00/wled00.ino line 21
0x402358f0: loop_wrapper() at C:\Users\Dirk\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp line 197
```

Aircoookie, 2020-08-20:
Vielen Dank, habe den PR gerade gemerged! Mir erschließt sich nicht wirklich der Sinn eines Publishes ohne Payload, aber da dieser Fall offenbar eintritt, bin ich froh dass wir durch einen simplen Check eine Crash-Quelle eliminieren können :)

closed 2020-12-26 by stale[bot]