wled/WLED #1087: WLED not connecting to home wifi

Mombonav, 2020-08-08:
I flashed my nodemcu ESP 8266 with WLED_0.10.0_ESO8266.bin via ESP home flasher. I joined the WLED_AP, and went into the config: wifi set up and entered my SSID and password, and a new mNDS. AP opens is set to No connection after boot. The I hit save and connect. I reset the nodemcu and on my computer I rejoin my home wifi and try to connect to wled via the mNDS in my safari browser however the board doesn't connect to my wifi. I have tried to connect via internet explorer on windows 10 and it still won't join my home wifi. The wled_AP always shows up in my list of available wifis. Client IP always shows Not connected. 

I tried unplugging the USB cable from the board and plugging it back in.  Previously I had the nodemcu connected to ESPhome via wifi and it worked fine. I tried to connect to my home wifi with the nodemcu next to the router and it still wouldn't connect. I am using an ASUS wireless router RT-AC86U. 

The router is dual band. I changed the name of each band and changed the password so that there were no special characters but it didn't resolve the issue. The wifi SSID has a space in the middle if that makes a difference. I placed the board next to the router to rule out signal strength and it still wouldn't connect.

It seems like wled never has the opportunity to join my home wifi, it just keeps defaulting to the wled_AP wifi.
This would indicate a SSID or password issue but I keep verifying and re-entering them to no avail.

I have looked over other entries relating to wifi but they haven't helped me solve this issue.

Thank you for your help!

wilson3682, 2020-08-08:
Hi Mombonav, I had a similar issue, the problem was when I tried to connect the NodeMCU ESP8266 to my 5G access point, when I changed it to connect to the 2.4G everything worked. I don't know if it's the same with you but I'm sharing my experience is case it helps.

Aircoookie, 2020-08-09:
Hi!
Sorry for the trouble connecting! Another user has reported that their phone auto inserted a space after the SSID, please make sure that isn't the case for you. You could also try setting a static IP in case the issue is related to DHCP (unlikely however).
After resetting the NodeMCU, does it still have the wifi information you entered saved? Just making sure it is not a flash issue :)

Mombonav, 2020-08-09:
There is no space and after resetting hte NodeMCU does still have the wifi info.
I set a static IP address and it connected fine to my wifi.
Not sure why the automatic connection isn't working.

sansillusion, 2020-08-09:
Ok, looks like your router dhcp server is having issues. Might be that you reached the limit of devices the DHCP server will assign. Most router are configured for 10 to 100 devices. Depending on router it could be a range setting or could be a limit in clients.

Wriman, 2020-08-10:
Hello! I have same problem, i try same things, but no connect...:(

StefanRu1, 2020-08-10:
Hi Wriman,
did you try with a static IP, matching the subnet of your router and not used by any other device on the network?
This should always work as long as the WLAN credentials you inserted are correct.

If this does not help please explain in detail, what you did and how you configured your ESP.

Mombonav, 2020-08-10:
So, my LEDs worked for a couple hours, then the pattern froze. When I reset the NodeMCU it would't rejoin my home wifi despite verifying the static IP, gateway, SSID and password. I even swapped in a brand new NodeMCU but it won't join the network either. Back to square one again. I have 24 clients on my network. When it did join my wifi the one time I had 26 clients then.  Any further suggestions?  Thanks

StefanRu1, 2020-08-10:
That sounds strange. Is it possible that you connect your NodeMCU via USB to your PC and open the serial console in Arduino Studio? You should be able to see what is going on and why the NodeMCU can not connect to your Network.

ellnic, 2020-08-10:
WiFi issues that seemed to resolve for no logical reason then freezing LEDs wreaks of a faulty board to me. I had some D1 mini clones a while back that displayed similar WiFi symptoms. Is your MCU from a reliable source? Does it work as expected when used for something else?

Mombonav, 2020-08-14:
Update.  I took my board (NodeMCU) to a friend who has an ASUS router (same as me just different model) and the board connected to his network fine. So not a faulty board. I took his D1 mini which was connecting to his network and brought it to my house. It would not connect.  I verified that the board wasn't connecting to my router by running a scan on my phone. Either the boards aren't seeing my wifi network or they are unable to join the network. I had previously changed the SSID to remove the space and changed the password all to no avail. Strangely, the board worked when I had ESPHome installed and it did connect once with WLED installed by static IP but only for 3 hours. It just seems like it isn't seeing the network for some reason. Either WLED software or my router settings. I have gone through the settings and nothing is jumping out at me. Extremely frustrating. Any hail Mary suggestions?

ellnic, 2020-08-14:
Hrm. Ok, so it's pointing to your router, as you suggest. 

A few things you can try (apologies if some of these are very basic, not trying to insult your intelligence):

1. Check your router firmware is up to date

2. Make sure all antenna are properly attached and tight

3. Make sure the antenna are pointing up or to a 45 degree angle [up left or up right]

5. Make sure you're using WPA2 Personal

6. Try channel 1 or 11 on the 2.4 band, don't use auto-hop 

7. Place the router away from obstructions and things that will interfere ie. cordless phones and microwaves, large metal objects

If these don't help, you may have a firmware issue or physical fault with your router. :-(

I once had an Asus laptop which I was very pleased to get. It was a ROG series and it looked great on paper. It was so riddled with firmware/UEFI issues that I was shocked to have paid the price it cost. I've had £400 Clevo's that ran smoother. The subwoofer and RAM also went faulty in it and Asus support was SHOCKING. They did fix the RAM and woofer under warranty, but it was a chore to get them to do it and they kept saying weird things like constantly referring to a stick of RAM as a "MEMORY CARD"... really strange behaviour. The UEFI bugs were never fixed. I also had to do the trackpad mod to stop it jumping around. In the end, the GPU went :-( A couple of years later I was persuaded to try an Asus router - one of the big beefy ones, I don't remember the model. It was awful. Low signal, constant disconnections etc. It went straight back to Amazon. So I vowed never to get an Asus product again. Point being - my experience with Asus has led me to believe that they have brilliant products on paper, but the product quality isn't there. You may have some kind of gremlin or hardware issue. I wouldn't bother contacting Asus about it though, they'll be useless. They'll say because a "normal computer" or phone can connect, it's not the router. If you even so much as mention NodeMCU their heads will explode. 

You may be able to find a 3rd party firmware for that router that should significantly improve it, and probably remove a ton of gremlins on the way. Not sure if you want to go down that route though. Or if you feel like it's time to move on, there are plenty of awesome routers supported by tomato and dd-wrt.

huggy-d1, 2020-08-14:
If your router is _smart_ and tries to combine both 2.4 and 5 GHz into a single SSID (for convenience), WLED will not connect, or it might connect some small fraction of the time (not very likely, just technically possible).
I found this out when my first WLED install on a D1 Mini refused connection.  A little online investigation suggested splitting the bands into unique SSIDs to allow IoT devices to connect.
That worked for me.

It might also be your router needs WLED not to enter sleep mode.  Some routers tolerate WLED's wifi sleep mode.  Other routers drop WLED from the network.

ellnic, 2020-08-14:
@huggy-d1 that's not it. See above, bands were separated. It's not the sleep setting either. The device doesn't connect at all, it's not that it goes to sleep and disconnects.

Mombonav, 2020-08-16:
Thanks to all for your assistance. I finally got the NodeMCU with WLED to connect to wifi.  I disabled sleep and set the control channel on the router to 11.  I think only changing the channel number mattered but I did them both at the same time. It connected without having to set a static IP.

closed 2020-08-16 by Mombonav

ellnic, 2020-08-16:
Glad you got it sorted. It's most likely the channel that did it, as you say. Channel hopping in routers is one of those features that looks good on paper but rarely causes anything but hassle. You can most likely allow the NodeMCU to sleep again by unchecking disable sleep. Now you have a solid channel on the router, I'd put money on it being able to connect. Happy WLEDing :-)

ianguerit, 2020-10-03:
Had the same issue with 0.10.2 (first time using WLED).
Switching the channel to be manual, channel 11 on my Asus router fixed it for me (left default settings, just set SSID and passphrase).

Super impressed by the effort that's gone in to WLED 👍

Mombonav, 2020-10-29:
Just after I got it working, it stopped working again. After two months I finally got it working again. This time I had to go into my router settings and disable smart connect, set channel to 11; set bandwidth to 20 and most importantly set wireless mode to legacy. Connection was immediate and stable.

smithjames0, 2020-11-29:
Hi, I don't want to open another thread as I have the same issue as described above. I have been running 0.10.2 on an ESP8266 for a couple of weeks without an issue, yesterday it would connect to my network but would not get an IP address. Nothing on the network had changed.
Today I have setup a 2.4G AP just for WLED on channel 11, but it refuses to connect and just goes into AP mode.

I don't understand how one day it works fine and the next it refuses to cooperate! 

My network;
Unifi 2 x AP lite's + Outdoor AP (2.4G)

Regards
James

ianguerit, 2020-11-29:
Check the channel bandwidth. Mine appeared to work for a bit, but then got unreliable. I found I had to set channel bandwidth to 20MHz, leaving it as either 20/40 or just 40 left it unreliable.

I don't think this is related to the WLED software specifically.

In summary, use 2.4GHz, control channel 11 and channel bandwidth 20MHz and you should be golden!

Also found these settings helped some smart plugs that disconnected every so often.

pepsonEL, 2020-11-30:
I also have a problem with my WLED 0.10.2. On GUI WLED wifi strange show info about 70-80% but my WLED sometime has delay with reaction on control.

Aircoookie, 2020-12-01:
@smithjames0 maybe https://github.com/Aircoookie/WLED/issues/1293#issuecomment-717929824 could help in your case

cjramseyer, 2020-12-31:
Make sure that you phone is not changing the case of the first letter of the SSID.  I was having issues getting it to connect to a UniFi network.  My phnoe calitalized the first letter of the SSID, which in my case is lowercase.  Once I corrected this and rebooted via the WLED AP GUI, it connected immediately

kiki3529-0, 2021-06-27:
Bonjour
Pourriez vous.m aider , je n'arrive pas à me connecter à wled-ap , je ne.trouve pas la solution.
Je suis chez BBox
La wemos mini d1 est flashé correctement
Le réseau wled-ap apparaît bien
Maïs impossible de se connecter
Pas d'espace 
Merci de votre aide
Mickaël

madc, 2021-08-19:
> If your router is _smart_ and tries to combine both 2.4 and 5 GHz into a single SSID (for convenience), WLED will not connect, or it might connect some small fraction of the time (not very likely, [...]

For us this also was an issue with Asus Smart Connect. I can't say, if it runs stable now, but blocking its MAC from using 5GHz made it connect properly.

ragu1953, 2021-10-17:
HELP! I usually connect to the ESP32 controller on a digquad board using the IP address but now I can't connect using either on an internet browser or the iphone app.     The IP address shows up on the router even when the digquad is not powered on. I am not techy so not sure where to go from here.

Mombonav, 2021-10-17:
I recently had this issue occur again to me. This time I was able to get the ESP32 to connect to wifi by not changing the mDNS address in the wifi setting. For that ESP32 as long as I keep the default value it will connect. As soon as I change it, it won't connect. For me, different boards need different solutions to connect it seems.

asmi242, 2022-07-20:
> Hi! Sorry for the trouble connecting! Another user has reported that their phone auto inserted a space after the SSID, please make sure that isn't the case for you. You could also try setting a [...]

Thanks for this. I imagine it would have taken me a long time to  figure this out without your comment...

bert269, 2023-02-02:
I had EXACTLY the same problem with my wifi on an ASUS AX11000 router. I had to reset my router to fact.defaults. Then I had to re-config all my IOT devices. I needed to re-configure the WLED devices and I had a heel of a time. Spend two full days flashing it over ands over. I was afraid I would wipe out the eeprom after so many flashed. Then I see in a previous post that the problem could be the router. I changed the 2.4GHz channel to 11 and boom! - the WLED connect just fine. What gives? Why is this a problem? All the other IOT devices seem to connect just fine all the time - is it the chip that is used on the ESP devices or something in the software?

bert269, 2023-02-03:
I spoke too soon. Problem is back with another ESP  unit. Cannot connect to wifi.
I have tried flashing it over and over, with the WEB interface as well as esptool.py.

HELP!

dotMortis, 2023-02-06:
I think you should open the issue again.
Last week I received 3 new esp8266 D1 Mini from AZDelivery.
All of them are experiencing the same network problems.

I am using an Asus RT-AX86U router where the 2.4Ghz SSID is separated from the 5Ghz SSID and only channels 1-11 are used.
The esp8266 are flashed with the Wled Web Installer, just like all my esp32 that I already have in use (no problems with esp32).
At first it took several tries to get the first network connection.
After the first connection worked, I was able to configure the esp via the browser.
So far so good, until I restarted the esp, after that they connect after about 5-10 min or never again.
When I look at the device list on my router I see the esp listed, and as soon as I try to connect to the esp web interface it disappears.

I have tried both the release and beta versions of wled .... all show the same problems with the esp8266.

msiemers, 2023-03-04:
Same Issues here. 2 new AZDelivery D1 Mini. No connection to wifi possible.

ShKlinkenberg, 2024-01-05:
I'm having similar issues on my esp32 board. After flashing the board and entering the wifi password I can visit the device on my local wifi network. This is a 2.4 and 5 wifi network with same SSID. After unplugging and connecting to other power source it does not connect to my wifi but offers the access point instead. Oh and nothing is connected to the GPIO ports.

I tried

- Creating separate SSID for 2.4 wifi
- version 0.14.0, 0.14.1-b1 and 0.14.1-b2

I'm new to the whole IoT space and esp32 / wled. but now I'm at a loss. Any ideas. Recompiling firmware sounds a bit daunting to me.

blazoncek, 2024-01-05:
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.

ourgangmn, 2024-03-08:
I have one ESP32 that has been a nightmare to get connected. It works as expected if connected to my laptop via a USB cable. I even put the board on a developer board and the power from the 5v power supply only is connected the Vin and GRD pins, bypassing the microUSB connection. That worked for a week. Now it refuses to connect to any wifi connection. I even separarated 2.4 and 5 GHz bands on my router, simplified the password, changed the channel to 11 and limited the band to 20 MHz. The absolute only way to connect wirelessly to  this device is via the AP that the board creates because if refuses to connect to wifi. And, the Wifi password has been entered correctly in the Wifi Config page. This board seems to be worthless as far as network connectivity goes.

ShKlinkenberg, 2024-03-09:
@ourgangmn 

For me this was my first experience with ESP32. It turned out it was a faulty board. I had a two pack, and the other one worked right of the bat. Wasted so much frustrated time on this. But let”s get of the dev forum, it is not software related.

ellnic, 2024-03-09:
@ourgangmn as @ShKlinkenberg says, this is probably a faulty board.

I don't always go for Lolin units depending on price, and I had a batch from a particular Ali seller that were all bad. The seller was reputable and I had ordered from them on and off for years - but things like this happen.

My symptoms were exactly as you describe - they would connect to the personal hotspot but not to the router. I suspect this was an antenna issue due to the proximity of my phone when setting up. 

The seller in my case was obviously aware that some were bad and refunded without quibble, and I continue to order from them. 

You did all the correct changes in your router, so communicate with the seller and hopefully get a refund for these and grab a different batch. I bet you'll be fine.