wled/WLED #1088: "Auto-calculate white channel" setting not being saved

shendriksza, 2020-08-08:
When I select **Auto-calculate white channel from RGB** in the LED settings and save, the setting is never stored. I always still have manual control over the white channel, so `rgbwMode` is definitely still set to `RGBW_MODE_MANUAL_ONLY`, confirmed by reloading the LED settings and the checkbox not being ticked.

I'm trying to debug by examining the HTTP POST we send to the device and it seems ticking the checkbox sends parameter `AW=0` while unticking it omits the parameter. Seems like that might be an issue considering we evaluate the parameter in `set.cpp` as `request->arg("AW").toInt();` which would probably always evaluate to 0.

So to fix it, I changed it to `request->hasArg("AW");` which successfully sets`rgbwMode` to `RGBW_MODE_AUTO_BRIGHTER` and the feature then works as expected! (although `RGBW_MODE_AUTO_ACCURATE` is probably the desired mode for this setting?) Unfortunately, the checkbox still does not populate to show the new setting. I'm not sure how it gets loaded into the HTML, seems there is some JavaScript involved

Hope someone can narrow down the intended use case and shed some light on this (chuckles), seems this feature was created here: #573 

**To Reproduce**
Erase ROM and flash, connect to WLED AP, tick `LEDs are 4-channel type (RGBW)` checkbox and then `Auto-calculate white channel from RGB` checkbox, and then save.

**Expected behavior**
`Auto-calculate white channel from RGB` checkbox setting is suppose to be save and white channel in RGBW strip should illuminate when picking a colour that contains white.

**WLED version**
 - Board: Wemos D1
 - Version 0.10.1
 - Format Binary, verified with a self-compiled

Thank you for your help!

Aircoookie, 2020-08-09:
Thank you for the report!
The checkbox had been replaced with a multi-choice dropdown menu before 0.10.0.
Unfortunately this change was not introduced into the `settings_leds.htm` file.
Since a script is now utilized to generate the final settings from those files, this reverted it to the older checkbox.
The ESP side code expects the dropdown, therefore the checkbox is not working correctly.
Will fix by adding the dropdown to the htm file again :)

Aircoookie, 2020-08-10:
Fixed!

closed 2020-08-10 by Aircoookie

shendriksza, 2020-08-11:
@Aircoookie That's amazing, thank you so much for all your effort! I'll be taking a look at your commit to better understand how this works :)

commit 5a4b7199 (2020-08-10): Fixed Auto White mode setting (fixes #1088)