wled/WLED #1094: YOu can't use WS8211S with the code

Stoppersocke00, 2020-08-11:
**Describe the bug**
A clear and concise description of what the bug is. Please quickly search existing issues first!

**To Reproduce**
Steps to reproduce the behavior, if consistently possible

**Expected behavior**
A clear and concise description of what you expected to happen.

**WLED version**
 - Board: [e.g. Wemos D1, ESP32 dev]
 - Version [e.g. 0.10.0, dev200603]
 - Format [e.g. Binary, self-compiled]

**Additional context**
Anything else you'd like to say about the problem?

Thank you for your help!

Aircoookie, 2020-08-12:
Hi!
WS2811 are compatible. Can you send a link to where you bought the LEDs from? How old are they? If they are older than 5 years, they might only work with half the data rate which would need to be compiled that way.

toto79, 2020-08-13:
for me it works with an ESP01 and NeoEsp8266Uart1Ws2811Method. e.g. in original NpbWrapper.h line 120-121
`define PIXELMETHOD NeoEsp8266Uart1Ws2811Method`
if you use an ESP32 then maybe NeoEsp32Rmt1Ws2811Method in Line xy is a solution.

here i found view which Methods are available: [https://git.htl-klu.at/5440121001/lucifermateriallager/-/blob/master/NeoPixelBus_by_Makuna/keywords.txt](url)

Stoppersocke00, 2020-08-17:
Hi!
Here i have the Link where I bought it:
https://www.tme.eu/en/details/hc-f5v-60l-60_20/light-sources-led-ribbon/worldsemi/hc-f5v-60l-60led-w-ip20/

They are not older than 5 Years and have the WS2811S Note the "S"
Controller inside.

toto79, 2020-08-18:
Did you tried the NeoEsp8266Uart1Ws2811Method ? Mostly it is a question regarding wiring or timing (and not explicit the age). Follow my link and you'll find two timings 800 khz and 400 khz. Sk6812, ws2812b, ws2815 should be compatible to 800khz, but the ws2811 is designed to use 400khz.  .... sometimes it works with 800, sometimes not ;)

Stoppersocke00, 2020-08-24:
No! I have not tried it yet, becuase I don't know how. Im new at this so can you Tell me whre I can change Method please?

toto79, 2020-08-25:
**Option A)**
you can give https://gitpod.io#https://github.com/Aircoookie/WLED/tree/master an try

run 1st: pip3 install -U platformio
run 2nd: platformio run -e env:type , e.g. platformio run -e d1_mini or platformio run -e esp01_1m_full (as defined in platformio.ini)

find your firmware.bin file in folder .pio/build/env:type, save as... and flash to your esp
good summary you'll find also at https://tasmota.github.io/docs/Gitpod/

**Option B)**
Install and use VSCode & PlatformIO local
https://www.youtube.com/watch?v=7axaXO8cFo8

For a clear and detailled description, you have to provide more information about your hardware. See missing entries in your 1st post, e.g. board. The following assumes a standard esp8266, like a Wemos D1 with LED connected to pin2 (=D4)
> change line 121 in file wled00/NpbWrapper.h to #define PIXELMETHOD NeoEsp8266Uart1Ws2811Method (instead of #define PIXELMETHOD NeoEsp8266Uart1Ws2813Method)
> compile and upload

closed 2020-12-31 by stale[bot]