wled/WLED #1095: NodeMCU with a 5V 3A PSU not lighting up a 150 LED WS2812b strip

ArifSy, 2020-08-15:
Hello all,
I initially set up my NodeMCU following DrZz's video using jumper wires to connect my ESP8266 to the strip, and using an external wall connected power supply to power the setup. I connected Vin to +5V on the LEDs, but accidentally switched the ground and data pins on the LED strip. Since it was not working, I reconnected it correctly with GND to GND, 5V to Vin, and data in to the D4 pin. My lights initially did not light up at all with either wiring, with just a very dim, barely visible light appearing on each of the LEDs. The WLED software works correctly, I can connect to the controller using my phone but changing the preset, or turning it on or off does not affect the LEDs at all. My psu is also connected as DrZz did, with the red lead connected to the positive end and the white lead to the negative end. I have also made sure that the directionality is correct.

For further information, I am using the BTF-lighting WS2812b IP65 5m strip with 30 LEDs/m (so 150 LEDs total).
Here's what I have tried to troubleshoot the problem, and in all cases the result was almost always the same:
- Re-flashed the controller.
- Re-flashed it but with version 0.9.1 instead of 0.10.0
- Power cycled, so turned it on and off.
- I tried powering the strip from the other end, and there was no change in behavior. 
- At this point I suspected that my psu was not powerful enough, so I tried it with a 5V 10A 50W psu as recommended by the manufacturer to no change.
- I re-connected all the wiring with different jumper wires, in case there was something wrong with the wires themselves.
- I read online that connecting to pin D2 might help, but it did not.
After I connected the D2 pin to the data in pin for the LEDs, one singular LED turns bright green, with all the others still inactive. After reconnecting the jumper wire from D2 to D4, the singular LED is still bright green.

Some specs:
 - Board: ESP8266 bi HiLetgo with 4 MB flash storage
 - Version: have tried both 0.10.0 and 0.9.1
 - Format: Binary
 - Lights: 5m strip, 30 LEDs/m, 150 LED BTF-lighting WS2812b strip

I would greatly appreciate any help, as I am unsure why my setup is working incorrectly and also because I am a novice with respect to circuitry and wiring.

Legsmaniac, 2020-08-15:
It sounds like you have blown your LEDs.
From experience,  it's possible that only the first LED is damaged.  Solution - cut off the first LED leaving you with 149 LEDs on the strip and try again, being very careful with your connections this time.  Chances are you'll find all is well. If not, cut a 2nd LED off and repeat the process until you find the remainder of the strip working correctly.

raoulteeuwen, 2020-08-15:
Could you share some pictures clearly showing how all wires run, so from your PSU to your LED strip, from your PSU to the nodeMCU and from the nodeMCU to the LED strip.

Aircoookie, 2020-08-15:
Hi, your LEDs probably are not broken since WS2812b luckily offers reverse voltage protection. The reason for the dim light could be the auto brightness limiter kicking in. Please check that you have set the `Maximum current` to the rating of your external power supply :)

ArifSy, 2020-08-15:
> It sounds like you have blown your LEDs.
> From experience, it's possible that only the first LED is damaged. Solution - cut off the first LED leaving you with 149 LEDs on the strip and try again, [...]

I wanted to do this step last, so I won't lose money if anything is wrong with the strip.

> 
> 
> Could you share some pictures clearly showing how all wires run, so from your PSU to your LED strip, from your PSU to the nodeMCU and from the nodeMCU to the LED strip.

[Here is an album with my setup](https://imgur.com/a/oOKtPZ3).

> 
> 
> Hi, your LEDs probably are not broken since WS2812b luckily offers reverse voltage protection. The reason for the dim light could be the auto brightness limiter kicking in. Please check that you have set the `Maximum current` to the rating of your external power supply :)

I have tried removing auto limiting brightness, and also setting maximum current to both 10000 mA and 3000 mA depending on what PSU I used.

raoulteeuwen, 2020-08-15:
As far as i can tell, looks fine, connection wize. Any chance you can try with another LED strip?

Legsmaniac, 2020-08-15:
> your LEDs probably are not broken since WS2812b luckily offers reverse voltage protection

Normally that should be the case but it's happened to me.

> I wanted to do this step last, so I won't lose money if anything is wrong with the strip.

If you cut, for example, 5 LEDs off the beginning of the strip, you can always solder them back on again if it turns out not to be the problem. 
Your wiring and connections looks fine so I really willing to bet it's your strip.  Unless you have a problem with your ESP or programming.  

Something else to try - again, in my experience, I've had a few ESP8266s where they freeze/lock up on power up with using pin 2.  You can test this by disconnecting pin 2, power up, then connect pin 2.  If it works, best bet is to compile with new pin set output.  I prefer pin 4 or 5.

Legsmaniac, 2020-08-15:
PS.......
As for the single bright green LED - that could be down to the fact you initially wired it incorrectly so you may need to cut past that LED to test the remaining strip.  Or it's possible that only that LED needs cutting out.  
OR - it's possible it's just a dodgy LED that will light up when powering the strip but will function normally when the WLED starts functioning. I had this with one LED on my matrix of 2040 LEDs.  It was fine but the fact that it kept coming on green annoyed me so I cut it out and soldered in a replacement.  It was easy enough and the strips are cheap enough after all.
The joys of cheap Chinese components, huh?  😉

emardee, 2020-08-17:
If you are wanting to test the strip, rather than cut it, you can probably use a wire to carefully inject the data from P2 (or whichever pin is your data), directly to the "Data in" on the strip inbetween LED1 and LED2 (instead of using the connector on the end). Should give the same effect as cutting (eg skip first LED) without damaging strip for possible return. You can try anywhere along the strip in same manner.

Sounds like strip isn't getting the data signals. I'd definitely double check pins. Some controllers use different pin numbering for the pins, so your P2 might be number P13 or something weird.... so you'd need to check pinout of your controller.

Is the LED strip wired directly from PSU, or drawing its power from the controller? Ideally you want any strip with more than a handful of LEDs to be wired direct to psu for power.

Legsmaniac, 2020-08-18:
> If you are wanting to test the strip, rather than cut it, you can probably use a wire to carefully inject the data

This will not always work, sadly. I have tried this myself many times and the affected faulty LED can lead to short circuiting so you need to bypass it completely. If you really don't want to cut, you can always try unsoldering the first joint. Strips are soldered together, usually every half metre so could always be soldered together again for return.

However, alas, unless I'm mistaken, looking at the videos you supplied, it looks like you bought IP67 waterproof strip that is covered in silicone so you have no choice but to melt/cut the covering in order to access the pads for the LEDs, in which case you will NOT be able to return it.  Off this is something you really need to do (naughty really) then I suggest you return it for replacement first then start again with a new strip.

Donpilkerton, 2020-08-28:
Not sure if this is going to help you out but I was having issues with my WS2812B leds lighting up as well when using an external 5v 10amp power supply, or when using usb. I am not sure what controller you are planning on using but I am using https://github.com/NimmLor/esp8266-fastled-iot-webserver and when specifying what pin I was using I had to input it as 4 not D4. I was not able to control my leds or have them power up until I made this change. I had a single led that would light up green or white randomly but doing this fixed my issue.

closed 2021-01-02 by stale[bot]