wled/WLED #1096: WLED not connecting to wifi when data pin is connected.

darkcloud784, 2020-08-15:
**Describe the bug**
I can get the NodeMCU to power on and connect to wifi when the datapin is not connected to D4. Once I plug in the datapin it won't connect. I've searched previous bugs and found a suggestion of putting a pullup resistor between 3.3v and data as the pin might be pulling low. This did not work unfortunately.

**To Reproduce**
Flash WLED to ESP8266 NodeMCU connect it to your wireless, plugin VIN and GND to live/GND wires on the strip. Plug data pin (D4/GPIO2) to data of strip. Power strip using 5v, 10A power adapter using barrel adapter. Watch as the nodemcu powers on but does not connect to wireless.

**Expected behavior**
Do all the above and have it connect to the wireless and LED's come one when clicking power.

**WLED version**
 - Board: NodeMCU ESP8266
 - Version 0.10.0
 - Format Binary

Thank you for your help!

raoulteeuwen, 2020-08-15:
How many LEDs? Are you powering your LED-strip over your NodeMCU? (i think one can read your text in multiple ways, just checking). I assume VIN & GND both run from your PSU to the NodeMCU, as well as from the PSU to the LED strip? I assume the NodeMCU and LED strip are sharing ground. Make sure you switch off all options in WLED, especially Alexa and mDNS (but the more you de-activate the better for testing none of those are interfering)

sansillusion, 2020-08-15:
I had a similar issue where I was connecting to the ground near the vin on
my board for some reason when connecting ground to opposite side of board
instead everything worked normally.

darkcloud784, 2020-08-15:
> I had a similar issue where I was connecting to the ground near the vin on my board for some reason when connecting ground to opposite side of board instead everything worked normally. Le sam. 15 août 2020 2 h 21 p.m., Raoul Teeuwen <notifications@github.com> a écrit :
> […](#)

I tried both grounding points on the board, no change.

> How many LEDs? Are you powering your LED-strip over your NodeMCU? (i think one can read your text in multiple ways, just checking). I assume VIN & GND both run from your PSU to the NodeMCU, as well [...]

Powering it the other way around, NodeMCU through the strip. I've tried with 10 LED's as well as 50, no change.

Aircoookie, 2020-08-16:
Hi, your issue is probably the same as the one you found previously with the pullup resistor as a fix. What resistor value did you try? (typically 3.3k or 4.7k work well). If nothing else works, the problem would likely be circumvented by switching to LED pin GPIO4 and using that binary :)

darkcloud784, 2020-08-17:
> Hi, your issue is probably the same as the one you found previously with the pullup resistor as a fix. What resistor value did you try? (typically 3.3k or 4.7k work well). If nothing else works, the problem would likely be circumvented by switching to LED pin GPIO4 and using that binary :)

I tried with a 4.7k ohm resistor. I'll try using GPIO4 and seeing if that works.

ilandllc, 2020-09-22:
@darkcloud784 - I seem to be having the exact same issue.  Did you find a resolution?

sansillusion, 2020-09-22:
Try to use the ground pin on the opposite side of the Vin pin. I had a similar issue where if I used the ground pin on the same side of the Vin pin while leds where connected  it would bootloop.

ilandllc, 2020-09-22:
@sansillusion - I turned off the power, swapped the ground pin to the other side, fired it back up and it worked.....for about a minute.  So close!  This is seriously close to driving me crazy.

Hicsvntspacedracones, 2020-10-27:
I am having the same issue also. I had checked the 12f and did catch a solder bridge from the gpio2 pad to the metallic cover of the 12f shorting to ground@Fixed that, but still same issue. I tried led pin 16 with nothing happening, so went back to main firmware on pin 2. I'll give gpio3 a shot, but as of right now, I get wifi until I connect to gpio2, then my lights come on but wifi drops and verified no shorts.

maupie1967, 2021-02-01:
I had the same issue with NodeMCU v3 board, this has a different pin layout and had to use the Vusb pin instead of Vin

closed 2021-06-08 by darkcloud784

tome9111991, 2022-04-05:
Hi i had same issue with Nodemcu v3

I swich to gipo 4 (D2) is working now

blazoncek, 2022-04-05:
GPIO4 is not recommended for LED counts above 30.

tome9111991, 2022-04-07:
> GPIO4 is not recommended for LED counts above 30.

Ok thanks.

I use now a 4.7k pull up between 3.3V and D4 (GPIO 2)
It works now.

lober-io, 2024-08-19:
I have exactly the same problem but I don’t get the pull-up working. Can someone draw or explain how to connect it correctly? 

I‘m also not sure if this will be the solution as i have the same issue on pin 4 with just 3 led :( 

The last time everything worked smoothly with the same components…