wled/WLED #1098: WLED not working with Sonoff Basic

utech-git, 2020-08-16:
Hello there.

I'm having trouble getting the WLED to work with a sonoff basic.

The LED strip commands doesnt work for change color/effect/brigthness. The only command that work is the power on/off. But I can hear the sonoff relay changing its state.

The wiring diagram is as follow:
[image]

Here's an example of the led strip when it's turned on... some of the LED keep changing colors randomly from time to time. It doesnt repond when I change the colors from the web UI
[image]

I am using this LED strip (WS2811): https://www.aliexpress.com/item/33027018662.html

I have tried using the [WLED_0.10.0_ESP8266_1M_ota.bin](https://github.com/Aircoookie/WLED/releases/download/v0.10.0/WLED_0.10.0_ESP8266_1M_ota.bin)  (using the GPIO2 from the sonoff, that is the soldered pad below the button). And I have also tried compiling the code and changing LEDPIN to 3 on NpbWrapper.h (GPIO3 is RX on sonoff).

My LED Strip is BGR, I tried changing it on the LED PREFERENCES page on the web UI but didn't work. I have the same led strip working ok with D1 Mini chip. But no luck with sonoff.

Am i missing something on the configuration to make it work with sonoff?

Thanks.

mjferro, 2020-08-16:
You need a common ground between the strip and the low voltage side of the sonoff.

utech-git, 2020-08-16:
And remove the ground between the power supply and the led strip?
Like this: 
[image]

I know I could just try the wiring from the image and see if that works instead of asking, but since I'm afraid i might break something I think it's better to ask first :)

Thanks

mjferro, 2020-08-16:
Keep the 12v PS to strip ground and add the esp to strip ground. The low voltage grounds should be connected together to have a common ground: esp8266 gnd, 12v PS gnd, & LED strip gnd all connected together.

utech-git, 2020-08-16:
Perfect!
Now almost everything seems to be working correctly.

I'll let the final wiring diagram here in case anyone stumble upon this thread in the future.

But now I am getting LED flickering. And once in a few seconds the LED change colors and then back to normal.
After changing the sonoff basic for a sonoff SV it is working correctly.

I really wanted to use the sonoff basic because with it I could turn on/off th 12v PS and it already has a case.

I don't think the problem is the LED strip nor that I should use the "[Skip First LED](https://github.com/Aircoookie/WLED/wiki/FAQ#my-leds-act-funny-and-flicker-randomly)" strategy described in the FAQ because it works with the sonoff SV.

I'll close the thread, but if anyone has any idea on how to eliminate the flickering i'd be glad to hear.

Thanks!
[image]

closed 2020-08-16 by utech-git

FulvioSpelta, 2021-03-30:
Hi, I've experienced the same behaviour you describe.
I'm testing now using sonoff basic V1 and using:

- a modified fw (the stock one downloaded in binary form wasn't stable) #1590 
- Erasing the chip before flashing #1841 
- a good (short) ground connection between the ESP ground and the led strip ground

I'll do some stress tests, currently I'm driving a 300 led 5V strip nicely

FulvioSpelta, 2021-03-31:
I confirm after a full day stress test that it's working with effect, NTP and MQTT enabled.

The firmware has been compiled using v 0.11.1 sources but using this job:
```
[env:esp01_1m_test]
board = esp01_1m
platform = espressif8266@1.8.0
platform_packages = ${common.platform_packages}
board_build.ldscript = ${common.ldscript_1m128k}
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags_esp8266} -D WLED_DISABLE_ALEXA -D WLED_DISABLE_BLYNK -D WLED_DISABLE_CRONIXIE
    -D WLED_DISABLE_HUESYNC -D WLED_DISABLE_WEBSOCKETS  -D WLED_DISABLE_OTA -D LEDPIN=3
```
See [this comment](https://github.com/Aircoookie/WLED/issues/1590#issuecomment-753332142) for full reference #1590 

I'm using LEDPIN=3 in order to use the RX pin as led driver (that is available on the pcb) avoiding the use of GPIO14 (the gpio available on pcb) that, in my test, let the system unstable.

Currently i'm using a 74HC245 ic as level translator.

bobbynobble, 2022-03-25:
> 

Hi @FulvioSpelta  I flashed a Sonoff Basic v1 with the T8 bin you kindly shared here... https://github.com/Aircoookie/WLED/issues/1590#issuecomment-753332142

All seems to work, many thanks. I now want to drive an apa102 strip with it which needs a clock and data connection. Will your bin work with that or if not could you give me any pointers how to do it? They are currently driven by an rpi zero w running Hyperion and I'm hoping to host Hyperion elsewhere and replace the rpi with WLED on the Basic so I can use the relay to power on and off the LED power supply.

Any help would be much appreciated as I'm totally confused at the moment.

Aircoookie, 2022-03-25:
@bobbynobble that binary is based on an old version of WLED that did not support configuration of strip type and pins.
I would recommend you flash the latest binary `WLED_0.13.1_ESP01.bin`, which allows you to set the APA102 type and pins you are using right in LED settings.

bobbynobble, 2022-03-25:
@Aircoookie many thanks for the reply, I did try that bin originally for exactly the reasons you mention but the WLED AP would never appear, then when I found the issues @FulvioSpelta had had with the same version of the device I had and how he fixed them, I tried that and it loaded first time.

If you could give me some idea what might be going wrong with the current bin and how I can fix it that would be much appreciated as I'd obviously prefer to run that if possible.

Aircoookie, 2022-03-25:
@bobbynobble thank you for letting me know, I've attempted compiling 0.13 with the old arduino core from @FulvioSpelta 's T8 test. Seems to work, but I haven't tested extensively, here is the binary: https://drive.google.com/file/d/1joU3ZmxjLTRr_4ZtWEjyZ4GFtK8186BO/view?usp=sharing

Built using this environment:
```ini
[env:esp01_1m_oldcore]
board = esp01_1m
platform = espressif8266@1.8.0
platform_packages = ${common.platform_packages}
board_build.ldscript = ${common.ldscript_1m128k}
upload_speed = 921600
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags_esp8266} -D WLED_DISABLE_ALEXA -D WLED_DISABLE_BLYNK
    -D WLED_DISABLE_HUESYNC -D WLED_DISABLE_OTA
lib_deps = ${esp8266.lib_deps}
monitor_filters = esp32_exception_decoder
```

bobbynobble, 2022-03-26:
@Aircoookie That's amazing, thank you so much for doing that. I'll flash it in the morning and let you know how it goes..

FulvioSpelta, 2022-03-26:
@Aircoookie Thanks a lot !

bobbynobble, 2022-03-26:
@Aircoookie Don't know what I'm doing/done wrong but the AP doesn't show up after powering it on but sometimes appears if you wait long enough. When it does, I can add the password which it accepts and then says it's getting an IP address, couple of seconds later it vanishes and won't come back unless it's rebooted and then only sometimes as before.

Any ideas?

FulvioSpelta, 2022-03-28:
@bobbynobble this is the behaviour that i've experienced with the old version.
Jist an hint: when you flash the firmware take care to wipe the eeprom  first.

bobbynobble, 2022-03-28:
@FulvioSpelta many thanks for the tip, the flasher does do an erase before writing but I guess this isn't enough. What would you recommend as the best way to do it?

bobbynobble, 2022-03-29:
Cheers, I'll give it a go. Hope I can get it to work as it seems the perfect solution and I have several V1s not currently used for anything.

bobbynobble, 2022-04-02:
@FulvioSpelta still no joy unfortunately, even tried esptool.py :(

FulvioSpelta, 2022-04-02:
I'm afraid, maybe it's time to consider to use the old version with ws28xx strip in order to use only 1 line.
My test shown that the V1 is a bit "sensible" and, for example, you can't use GPIO14 (i suppose that the driver code is too heavy) and i've had good result only using RX pin.

bobbynobble, 2022-04-02:
Think it's better (cheaper!) just to put tasmota back on it and get an ESP32 for WLED. I can still automate power with the Sonoff through Home Assistant, just not as neat as it could've been.

@Aircoookie many thanks again for trying to get it working for me.

mirko, 2022-12-03:
> @Aircoookie Don't know what I'm doing/done wrong but the AP doesn't show up after powering it on but sometimes appears if you wait long enough. When it does, I can add the password which it accepts and then says it's getting an IP address, couple of seconds later it vanishes and won't come back unless it's rebooted and then only sometimes as before.
> 
> Have tried reflashing several times but makes no difference.
> 
> Any ideas?

This is exactly the behaviour I'm experiencing - with all my Sonoff Basic (R1) devices, incl. the published T8 image. And/But Tasmota works fine here, too on them.
Anybody any idea, *why* this is happening?
Is WLED just not working on ESP8266 anymore - because of.. yeah, again, why actually?
It doesn't seem like it's crashing and even if so, why does it seem to get farther sometimes than at others?
It feels painfully erratic to me and I'd love to get have some more technical explanation about this (next to a working WLED firmware for those poor Sonoff Basics here :))