wled/WLED #1102: WLED IFTTT integration

MakeThat, 2020-08-21:
First off, I love WLED and believe that this project has been amazing. I have started to make some little lights and things with WLED and would like to integrate it into google home as a standalone service. I know that there are Blynk and Hue allow for basic controls of the light, and I was wondering if there was a way that could allow for the sun to "Pull" data from the public-facing server, so the sign doesn't have to be exposed to the public network. 

I think this would pretty much be a web server that has a custom generated tag that would be an identifier for the WLED device. Then there would be a second code that is created similar to an API private key that would provide basic authentication. This private key could also be regenerated to allow for public side connectors to no longer work, but the WLED lights to have no settings needed to be changed.  So an example query to this service would be `_WLEDPublicConnector_.com/_ServerGeneratedSign_/&U=_PrivateKey_`

Then, In theory, the user would be able to use the HTTP API or the JSON API to communicate to this server, then the sign would be able to query the public server by using a JSON request, with a URL that looks similar to this.  The private key wouldn't have to be included in this query because there is no harm to have someone else use your lighting settings. `_WLEDPublicConnector_.com/Data/_ServerGeneratedSign_/`

On the WLED side, if it had a text box input that allowed you to put in the ServerGeneratedSign code and update the polling rate, that would be great. Then if it a very basic JSON parser were created, it would work great. I feel like the only values that need to be passed is if the sign is on, the brightness, effect number, speed, intensity, and R, G, B values. Then once the values are parsed, it just updates the light's current state. 

For security, I feel that this is pretty secure as is, and this idea is used in many API like applications, the only problem is rate-limiting, it would be easy to identify an Address that is trying to brute force a private key, but harder to determine what would be an abnormal update amount. There is also the problem of someone trying to scan for "working" Sign codes by pulling data from all of the combinations. I think this could be solved by having a script that can generate some randomized responses that include possible values, so there is no "default" unused API value. I'm not a security expert but I believe that this would work pretty well, and suggestions are more than welcome. 

I haven't looked at any alternative yet; if someone has a different idea or if there is a new, more secure process that is used now, feel free to share it.
This, I feel like would be a great feature that isn't very resource-intensive and would greatly benefit the community overall. It could be implemented as a free service or a software package that is downloadable and able to be self-hosted. An easy to use Web configuration tool would be great, but a text file that can be edited would also be acceptable. 

Any suggestions would be greatly appreciated.

tronicsmasta, 2020-11-28:
Got IFTTT working with Google Assistant. Not exactly what you are lookin for but might help someone else who is looking. 
https://www.instructables.com/WLED-on-ESP8266-IFTTT-Google-Assistant/

Legsmaniac, 2020-11-28:
Only problem is that IFTTT now only gives you the 3 routines for free. After that, they charge you a monthly subscription which, IMO, isn't cheap. IFTTT is good though and very simple to use.

billyjbryant, 2020-12-24:
In order to create a new Integration (Applet) for IFTTT you have to have a Paid IFTTT Plan. This makes this costly as the minimum plan to have published applets that are available for all IFTTT users to leverage is $199USD/year. 

The alternative might be, If you are using WLED as a Hue light, then you can use Hue's existing integration to control the LEDs but this would probably limit the use case to simple color changes, so your Sign example wouldn't really work this way.