wled/WLED #1106: Set current preset cycle setting as boot default does nothing

feraldruid22, 2020-08-23:
Hello,

I running the latest WLED 0.10.0 on an ESP8266 D1 mini.
I only using one single effect the lighthouse with cyan color.

What I would like to save this in case there is a power outage and the device always load this automatically on boot.
The only option I could find related to this in the settings is what I mentioned:

**Set current preset cycle setting as boot default:** 

I have also tried to enter preset numbers I don't know if they are supposed to be the same as effect numbers but no effect.

At reboot I always get orange solid light it does not restore anything.

karen8000, 2020-08-23:
Friend, you need to save the effect to the cell and in the settings to choose the default cell number.

Aircoookie, 2020-08-25:
Yep, you can just save it to one of the 16 slots in the Favorites tab and then set that slot number as the boot preset in LED settings!

closed 2020-08-30 by feraldruid22