wled/WLED #1107: WS2801 on ESP32

stshome2, 2020-08-23:
My WS2801 works fine on ESP8266 and WLED_0.x.x_ESP8266_ws2801.bin 

After change to ESP32 WROOM32 and WLED_0.x.x_ESP32.bin I can reach the AP and modify the settings but the LEDs are not "working".
I used IO02 and IO00 for data and clock as used  on ESP8266.
Is a special bin file needed for WS2801 or is it a bug ?
Best regards
Mike

huggy-d1, 2020-08-24:
This might help you greatly!  https://github.com/Aircoookie/WLED/wiki

stshome2, 2020-08-24:
No details found about how to solve the issue. 
Tried to compile a new version with uncommented ws2801 line.
Unfortunately not successful because of errors linked to mqtt.
Other idea ?

Aircoookie, 2020-08-25:
@stshome2 I recommend compiling with Platform IO, it will take care of all the library dependancies for you!

Here is a binary for ESP32 APA102 I compiled for you though:
https://drive.google.com/file/d/11ngCNZkkiR7xlRa2PmiEHxZldFR_b_rP/view?usp=sharing

stshome2, 2020-08-25:
Thank you for recommendation and great support with the file. Will test it asap in next days and feedback.

stshome2, 2020-08-26:
At the end I compiled a new file for WS 2801 with changed clock port (GPIO 17) and it work.
Only one difference vs first trial with 8266 and WS 2801.
8266 right colors with „RGB“
Esp32 right colors with „BRG“

pattyland, 2020-09-24:
Hi, I'm desperate to get my WS2801 with WLED running on the ESP32. After I didn't even understand that the normal build doesn't work with WS2801, I couldn't find which line I have to comment out. If I add `-D USE_WS2801` to the build_flags of esp32dev, the build will not work anymore.

@Aircoookie: Would be so kind providing ESP32 Builds for the WS2801? I still have a few meters left 🙈 

@stshome2: I was wondering about that GPIO 0 too... Would you mind sharing your working build?

stshome2, 2020-11-28:
@pattyland, 
sorry, was not here for some time.
I compiled my own file for ESP32.
Un-commended only the 2801 line in Led selection and changed the clock pin to GPIO17.  Just made it again once more for a Christmas decoration.
Unfortunately, no idea why the standard file doesn‘t work.
If you like I can upload the .bin file for Esp32 devboard.
From own experience: be aware about the direction of the stripe. ;-)

pattyland, 2020-12-02:
@stshome2 I'm not quite sure but in version 0.11.0 there is new binary called "WLED_0.11.0_ESP32_APA102.bin"
Is this the right file for WS2801?

stshome2, 2020-12-05:
@pattyland, the working principle is the same for WS2801 ans APA102. APA102 -the newer type can work faster than WS2801 that maybe results in different code. In the "NpbWrapper.h" a different PIXELMETHOD is defined.
For this reason, I would assume that your mentioned bin-file will not or not perfect work with WS2801.

moesfeld, 2020-12-22:
@stshome2 Sorry to dig this up again. I compiled WS2801 for ESP32 from the latest master and everything works fine on the software side of things. But unfortunately I'm not able to control more than 7 LEDs reliably. I compiled a simple fastled scatch and it seems to be an issue there as well. An ESP8266 can control the same String with 54 LEDs just fine. As far as I know the WS2801 amlpifies and flattens the CLK and DATA after each LED so I don't think it is voltage related. Measurements showed the same on ESP8266 and ESP32.
I think the frequency of hardware SPI is to high on the ESP32 over the bitbanged output on the ESP8266. Did you experience similar effects?

closed 2021-06-16 by stale[bot]