wled/WLED #1110: Addressing individual LEDs with MQTT

gshiftlabs-dburke, 2020-08-24:
I have WLED installed on a D1 mini with an 8x8 LED panel and MQTT working..

My goal is to build an pixel board which can display custom made pixel graphics in an 8x8 frame.   

1) Is it possible to set each pixel individually or via MQTT?
2) could this be done via a Custom Feature (does it have the ability to get some sort of param (pixel color array) from MQTT? (http api?)

my goal would be to move the complex logic of  building the pixel array to behind an MQTT server using node-red..   I can build an HTTP upload from an image then reduce / translate to a pixel array and send to the pixel board via mqtt.

huggy-d1, 2020-08-24:
Using E1.31 protocol, you can control each LED as you see fit.
Or, you can create a custom usermod to enhance the JSON API and/or MQTT API so it can.
However, speedy JSON/API or MQTT updates may add risk of crashing WLED.

gshiftlabs-dburke, 2020-08-24:
I will give the E1.31  a try.  Thanks!

Love the work you've done so far.!!

gshiftlabs-dburke, 2020-08-24:
Just to confirm (as I've never used E1.31 or Art-Net)   those are full network protocols?  they can't run over something like MQTT?

gshiftlabs-dburke, 2020-08-24:
Okay:   Sorry for this turning into a general Q/A, but I think I have this figured out and just need a few pointers on a few things..

it looks like I'm going to have to write my own custom user mod, but I'm unsure as to a few things
1) I'm assuming that I will have to use the htmlapi over mqtt (as I want to use mqtt) 
2) I won't need a customer topic, as the htmlapi already works
3) I need to pass the individual rgb string over the htmlapi so is this extendable?  is there a field which I can use and get access to its value in a custom user mod?

Aircoookie, 2020-08-25:
Indeed, right now unfortunately setting individual LEDs is not supported via MQTT, HTTP or JSON APIs (with JSON, segments are possible, but not individual control).
WLED does support this via E1.31, Art-Net or TPM2.net, and while I agree that setting individual LEDs at a fast rate via MQTT or HTTP is not a good idea due to TCP overhead, only sending them periodically (like in your 8x8 image frame use case) would be a nice feature. I'll be looking into getting this added to the JSON API (which runs over HTTP which is also supported by node-red).
Furthermore, JSON API support over MQTT is also planned (but there are a few memory-related issues, the reason it is not yet added)

Aircoookie, 2020-09-09:
> It sounds like the JSON API was upgraded to support individual LED control?

Yes, it was! We are still lacking a good way to convert images to the required JSON format though 😅 
Also the individually set LEDs will not persist after a power off (of the LEDs, not the entire ESP) and will revert to the effect mode.
Working on storing presets in filesystem right now which will also allow storing individual LED presets!

gshiftlabs-dburke, 2020-09-09:
why not use something like persistent messages in MQTT, that could be one way of having "storage" while not using memory on the device.

Aircoookie, 2020-09-10:
Once the MQTT interface supports the JSON API, individual LEDs are indeed a good use case for retained messages!

blazoncek, 2022-02-21:
MQTT now (since 0.11) supports JSON API so `{"seg":[{"id":0,"i":[....]},{...}]}` is possible to control individual pixels.
https://github.com/Aircoookie/WLED/commit/560f72a320f5a3c75f649c82c022dfdac4f339bc
Closing.

closed 2022-02-21 by blazoncek