wled/WLED #1111: ESP 8266 connects to phone through app but does not change lights whenever I press the buttons

FortLightning, 2020-08-24:
**Describe the bug**
A clear and concise description of what the bug is. Please quickly search existing issues first!

**To Reproduce**
Steps to reproduce the behavior, if consistently possible

**Expected behavior**
A clear and concise description of what you expected to happen.

**WLED version**
 - Board: [e.g. Wemos D1, ESP32 dev]
 - Version [e.g. 0.10.0, dev200603]
 - Format [e.g. Binary, self-compiled]

**Additional context**
Anything else you'd like to say about the problem?

Thank you for your help!

FortLightning, 2020-08-24:
I have all the wiring and programming done correctly. The lights come on but when I try to change the colors through the app or the computer nothing happens. It says that it is connected but nothing changes.

Aircoookie, 2020-08-25:
Hi!
What color do your lights illuminate in? Do all of them turn on? Does the master brightness slider or the on/off button work?
This might give us a hint at what is wrong.
In LED settings, make sure you have chosen the correct LED count and set the max. current to the rating of your power supply.

FortLightning, 2020-08-26:
Hey! The lights are green, blue, red, yellow, white, pink. They do not all turn on either. And the slider does not work aswell. I have adjusted the led light count but never the other other thing.

raoulteeuwen, 2020-08-26:
Sounds like something might be wrong with the data connection.

What LED strip exactly do you have?
What are the specs of your PSU?
Can you share a picture of your wiring?

FortLightning, 2020-08-28:
[image]

[image]
[image]

raoulteeuwen, 2020-08-29:
Looks good to me. What controller is that exactly? Data wire is on D4? What bin exactly did you flash to the controller?

FortLightning, 2020-08-29:
[image]
[image]

The 8266 bin

And yes the wire is on d4

Aircoookie, 2020-08-29:
Wait, is that 12v power supply attached directly to the LEDs?
In that case your LEDs are likely toast now :( The listing says they are 5 volt (most WS2812b are).
The NodeMCU typically also breaks at more than 6.5v input voltage, you are lucky if it survived.

FortLightning, 2020-08-29:
So what you are saying is that the issue is my power supply?

FortLightning, 2020-08-29:
Yes it is attached directly

FortLightning, 2020-08-29:
Could you send me what power supply i should get then for 150 leds please. And a strip I should get

FortLightning, 2020-08-29:
Also, the light still comes on on the nodemcu but could it still be broken you think?

Aircoookie, 2020-08-29:
Yes, this power supply has the wrong voltage for these LEDs.
You can either use LEDs made for 12v (WS2815) with this power supply (but you'd need a step down voltage converter for the nodemcu!) or what I recommend is to get the LEDs you have again (yours are 99% broken because of the overvoltage) and get a 5 volt power supply with at least 3 amps.

The nodemcu is not broken yet but disconnect it from the circuit immediately as it could blow at any point.

FortLightning, 2020-08-29:
Okay! Thank you! So is it fine if I get a 4amp or 5 amp power supply?

FortLightning, 2020-08-29:
I can't find any 3 amps

Aircoookie, 2020-08-29:
That is good, 3 amp is the minimum. More will give you more headway. Just make sure it is 5 volts :D

FortLightning, 2020-08-29:
Thank you! I will try everything you said and get back to you!

FortLightning, 2020-08-29:
[image]
[image]

I ordered these two. Will that work?

Aircoookie, 2020-08-29:
Yep, that should work!

FortLightning, 2020-08-29:
Okay, I'll let you know tomorrow!

FortLightning, 2020-08-31:
It worked! Thank you!

closed 2020-09-01 by Aircoookie