wled/WLED #1113: Corrupt heap reboots DOIT ESP32 DEVKIT V1

LordBoos, 2020-08-25:
**Describe the bug**
At whatever action I take in the web UI or from external source (Hyperion) I get corrupt heap error and ESP32 reboots. Reboots are 100% reproducible on all my 3 different ESP32. I use one of the 3 ESP32 for different usage (monitoring Mi Flora Plant sensors with ESPHome.io) without problems, so it looks like the problem is with WLED.
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
⸮Ada
dhcps: send_offer>>udp_sendto result 0
CORRUPT HEAP: Bad head at 0x3ffd592c. Expected 0xabba1234 got 0x3ffd5a68
assertion "head != NULL" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x4013542b on core 0

Backtrace: 0x4008d948:0x3ffb5e20 0x4008db79:0x3ffb5e40 0x4013542b:0x3ffb5e60 0x4008d5bd:0x3ffb5e90 0x400850de:0x3ffb5eb0 0x400858b5:0x3ffb5ed0 0x4000bec7:0x3ffb5ef0 0x4014e7df:0x3ffb5f10 0x4014e863:0x3ffb5f30 0x4014e8ab:0x3ffb5f50 0x4014f92a:0x3ffb5f70 0x4014f992:0x3ffb5f90 0x400f32ad:0x3ffb5fb0 0x40148ddc:0x3ffb5fd0 0x4008a091:0x3ffb6000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Ada
CORRUPT HEAP: Bad head at 0x3ffd51a0. Expected 0xabba1234 got 0x3ffd52f8
assertion "head != NULL" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c", line 214, function: multi_heap_free
abort() was called at PC 0x4013542b on core 0

Backtrace: 0x4008d948:0x3ffb5c70 0x4008db79:0x3ffb5c90 0x4013542b:0x3ffb5cb0 0x4008d5bd:0x3ffb5ce0 0x400850de:0x3ffb5d00 0x400858b5:0x3ffb5d20 0x4000bec7:0x3ffb5d40 0x4014e7df:0x3ffb5d60 0x4014e863:0x3ffb5d80 0x4014e8ab:0x3ffb5da0 0x4014f92a:0x3ffb5dc0 0x4014f992:0x3ffb5de0 0x400f32ad:0x3ffb5e00 0x40148ddc:0x3ffb5e20 0x4008a091:0x3ffb5e50

Rebooting...

```

**To Reproduce**
* Flash WLED
* Connect to wifi
* Press whatever button in the UI
I'am able to connect to my home wifi, because the configuration in the web UI is actually saved even when the ESP32 reboots, but after that any change to LED causes error and reboot. Turn-on -> Reboot, change color -> reboot...

**Expected behavior**
No reboots should happen.

**WLED version**
 - Board: https://docs.platformio.org/en/latest/boards/espressif32/esp32doit-devkit-v1.html
 - Version: 0.10.0
 - Format: Binary - WLED_0.10.0_ESP32.bin

**Additional context**
On my ESP8266, flashing WLED_0.10.0_ESP8266.bin works as expected without errors. I only have a problem with ESP32.

Thank you for your help!

Cuacco, 2020-08-25:
Hi
I am facing similar problems with esp32 (not with ESP8266) when I connect more of 700 LED, the esp32 starts to fail, freezing all LED in only one color (solid) after LED number 100, I must power off to reboot esp32 and begin again. 
I am not sure if I can connect usb to monitor serial port and obtain logs while I have connect it to power supply,

LordBoos, 2020-08-25:
My issue occurs even if zero LEDs at all are connected.

Kashyapkoshti, 2020-08-28:
rst:0x10 (RTCWDT_RTC_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 

I am getting this error in my esp32 devkit v1
How to solve it?
I am using arduino IDE for programming the esp32

Zebble, 2020-12-04:
Having a very similar issue with 0.11.0 on a devkit v1 clone.  With some persistence, I finally got the wifi config in and once it connects to my internal network it's stable.  Seems to only be a problem when the module is acting as it's own AP.  My gut says it's a power problem, so I may try on another module and with a better USB power supply or with an external power supply.

Kashyapkoshti, 2020-12-05:
@Zebble i used for 3-4 months and there was no problem but suddenly it shows this error and i tried with multiple supply but it doesn't run code and i can't upload New code.

closed 2021-06-16 by stale[bot]