wled/WLED #1114: Dynamic fade effect

benblb, 2020-08-25:
**tl;dr is bold below**

I went through all of the effects and couldn't seem to figure out how to do the following, but apologies if it's possible today.

Essentially I would like to see a way to have each group in a strip that uses the grouping feature (i.e. 100 LEDs on the strip in 10 groups of 10 pixels) fade between random RGB colors.

**The "Dynamic" effect would be perfect since it seems each section is a random color, but it jumps from color to color rather than fading. What I'm looking for is no predictable pattern. Essentially totally randomize each color and fade from one to the next.**

If I'm being really needy it would also be cool to have a dynamic fade effect where the different pixels change at different times.

I am currently in the process of building a Nanoleaf clone using WLED and this is my ideal go-to look since each light panel would appear to be behaving on its own. Thanks for reading and considering, and I'm a huge fan of your platform!

benblb, 2020-08-25:
My mistake, dynamic already has randomized times and is perfect! I downgrade my full ask to a fade of exactly how dynamic is today. 

PS thanks for getting eyes on this so quickly!

benblb, 2020-08-25:
It's been a while since I've done C++, but I took a stab at making dynamic fade which I pasted below. I hacked together the fade effect and the dynamic effect, so hopefully I didn't mess it up too badly! 

I didn't update the .h file (I assume this is just one line) or the UI. Not sure if the UI is automatically updated or not. 

`/*
 * Lights every LED in a random color. Changes all LED at the same time
// * to new random colors.
 */
uint16_t WS2812FX::mode_dynamic_fade(void) {
  if (!SEGENV.allocateData(SEGLEN)) return mode_static(); //allocation failed

  if(SEGENV.call == 0) {
	for (uint16_t i = 0; i < SEGLEN; i++) SEGENV.data[i] = random8();
  }

  uint32_t cycleTime = 50 + (255 - SEGMENT.speed)*15;
  uint32_t it = now / cycleTime;
  uint16_t fadedur = (cycleTime * SEGMENT.intensity) >> 8;
  
  uint32_t fade = 255;
  if (fadedur) {
    fade = (rem * 255) / fadedur;
    if (fade > 255) fade = 255;
  }

  if (it != SEGENV.step && SEGMENT.speed != 0) //new color
  {
	for (uint16_t i = 0; i < SEGLEN; i++) {
		SEGENV.aux[i] = SEGENV.data[i];
	  if (random8() <= SEGMENT.intensity) SEGENV.data[i] = random8();
	}
	SEGENV.step = it;
  }

  for (uint16_t i = 0; i < SEGLEN; i++) {
	fill(color_blend(color_wheel(SEGENV.data[i]), color_wheel(SEGENV.aux[i]), fade));
  }
  return FRAMETIME;
}`

Aircoookie, 2020-08-26:
Looks quite good already!
Did you get it to work on the LEDs already?

In any case, getting this effect in will definitely be cool!

benblb, 2020-08-26:
Thanks! Unfortunately my recent C++ experience doesn't go quite that far :p 

I drafted the above code but haven't been able to test. Do you see any obvious mistakes? I did my best to simply make the dynamic effect fade, any obvious mistakes that you can see? 

Out of curiosity, what would it take (assuming the above code is correct... it really should be tested lol) to add the effect to WLED? Is it just FX.cpp and FX.h, or do more changes need to be made?

commit a91d993c (2020-12-22): Dynamic fade effect (#1550)

* New Effect "TV Simulator"

based on "Fake TV Light for Engineers" by Phillip Burgess https://learn.adafruit.com/fake-tv-light-for-engineers/arduino-sketch

* removed some not-used functions 

...from my 1st attempt with "Phoney TV" - but this one did not look good.

* Dynamic Effect extended with a "smooth" variant

...to close #1114 / the "Dynamice fade effect" in the projects section
closed 2020-12-22 by Aircoookie