wled/WLED #1115: DMX start address for each segment on Artnet/SACN

abyssdj, 2020-08-25:
It would be handy to be able to set the starting channels (when controlled by DMX, Artnet, SACN, whatever) for each segment. As it is currently, the strips will "listen" based on the DMX start address in sync settings. If that is set at 1, the first LED will occupy channels 1-3 (for RGB), then the second 4-6, and so on. When segmenting, each segment still listens based on the led number. For example, a segment starting at LED 4 would listen from DMX channel 10.

It would be very handy to be able to set each segments start address in addition to the LED range it occupies, meaning that multiple segments (think 4x physical strips linked by cable from one ESP, for example) could be set to respond as required. In this case, 4x physical strips which may have 60 LEDs each, but all listening from DMX channel 1. The end result (to the eye) would be 4x strips doing the same thing, and would use less channels in software, save extra programming, etc.

Aircoookie, 2020-09-02:
I agree that this would be cool to have (right now the realtime mode ignores segments and just uses the entire LED strip), alongside it I might also be able to add the capability of running WLED effects on one segment while receiving a stream on a different segment!

abyssdj, 2020-09-02:
Yes, it would be very useful from a design standpoint. FX and a live stream may work, but would that not affect performance somewhat? I sometimes find that one of my units lags behind by a few seconds when on a live stream.

smalljoshua, 2020-12-20:
As an expansion of this, can sACN effect mode be applied over multiple segments?

swiss6th, 2021-02-28:
I'd also love to see DMX settings per segment, as I'm looking at using WLED in some stage prop designs. I have limited address space in my theater, and my current system only supports 1 universe. I can't really afford to map individual pixels into this universe, but with addressable segments, the channel count would be manageable.

visigoth, 2021-08-14:
slightly different from the OP, being able to set different DMX start universes for each segment would be helpful. the DMX controller software i'm using does not allow me to specify a different DMX start address for each fixture (only a different universe). afaict, WLED will simply increment the DMX universes as addresses roll over.

Jack079, 2022-10-08:
> I agree that this would be cool to have (right now the realtime mode ignores segments and just uses the entire LED strip), alongside it I might also be able to add the capability of running WLED effects on one segment while receiving a stream on a different segment!

Is this already possible? I would like to controll segments of 16 LEDs as one rbg fixture over artnet. I think this would be a better option than adding extra artnet universes to the control software

lorandnagy, 2022-10-25:
I would be also interested in this topic. I am here because of pioneer one dmx universe :)

deggle, 2022-11-08:
Just came across this - it's something I've requested and even put up a bounty for (I think it's basically the same feature as you're describing). I've just noted it seems to have been actioned - I will be checking this feature when I get chance...

https://github.com/Aircoookie/WLED/issues/2325

mxklb, 2022-11-20:
Please test PR https://github.com/Aircoookie/WLED/pull/2891 to control segments by sending DMX.