wled/WLED #1116: WLED adding support for MSI Mystic Light

lubbertkramer, 2020-08-25:
**Describe the solution you'd like**
With aRGB headers on my motherboard and support for Mystic light so my computer is matching all aRGB (fans, cpu, motherboard, gpu) into the same effect / color. I would like to expand the MSI Mystic Light to the WLED combined with ledstrips around my desktop/behind my screen 

MSI Mystic light
https://www.msi.com/Landing/mystic-light-rgb-gaming-pc/

huggy-d1, 2020-08-26:
Perhaps use the MSI Mystic Light SDK and control the LEDs around the desktop and behind the monitor (screen)?
Is there a protocol or API available that WLED can easily adopt, or does the MSI ML SDK already support a protocol available on WLED - or can easily be made to support?

NeariX67, 2020-08-28:
It takes a lot to implement a feature like you described. You need to code a desktop application that polls the current WLED Status (effects and colors etc) and implement it via mystic light. That means: EVERY effect, palette etc would need to be re-developed on a windows platform.
Problems with this idea:
It will become outdated very fast<br>
You need to detect individual hardware configurations
JSON, HTTP, E1.31.... protocol implementation etc...

This is not something you want to develop.

huggy-d1, 2020-09-03:
> It takes a lot to implement a feature like you described. You need to code a desktop application that polls the current WLED Status (effects and colors etc) and implement it via mystic light. That means: EVERY effect, palette etc would need to be re-developed on a windows platform.
> Problems with this idea:
> It will become outdated very fast
> You need to detect individual hardware configurations
> JSON, HTTP, E1.31.... protocol implementation etc...
> 
> This is not something you want to develop.

You mean like this difficult: https://github.com/FreekDS/MSI-Mystic-Light-Controller

sibest19, 2021-01-07:
A Windows desktop application could get the “LED settings” from the Mystic Light SDK and broadcast them via udp, taking advantage of the sync capabilities already available in WLED. Could this be a feasible solution?