wled/WLED #1119: Lights not working

superbond05, 2020-08-28:
Hello
I am new to this site and the world of programmable Led strips or any other Led lights. 
 I received my sw2811 led strip lights (18 15ft strips) today. Took one roll out and hooked it up to my 12v 60 amp power supply. I ran 12v to a step down voltage box that put out 5v. I flashed my esp32 chip with the latest version of wled. I could see the controller on my wled app. I plugged in my power supply and nothing! No lights and no control with the app. The all of a sudden LIGHTS! They all lite up a bright light blue but still could not control them. I tried a different roll of lights. They flickered muilti colours for a Mili-second. Then they went out. Supper frustrated! Ready to send everything back!  Please help me understand the Led world!
My goal it to led all our windows and eavestrough

**WLED version**
 - Board: [e.g. Wemos D1, ESP32 dev]
 - Version [e.g. 0.10.0, dev200603]
 - Format [e.g. Binary, self-compiled]

**Additional context**
Anything else you'd like to say about the problem?

Thank you for your help!

superbond05, 2020-08-28:
One more thing. When the esp32 chip is plugged into my laptop it has the wifi blue light flashing. If it unplug it and hook it up to the down verter 5v. With data on D4 it is powered up but no flashing blue. Which I a soon is the wifi light. Don’t get it!

raoulteeuwen, 2020-08-28:
O.k. Not sure how much time you've already invested in trying to grasp the basics. LEDs are not rocket science, but as with building rockets, you do need to invest some time to read stuff and/or watch some video's. My blog at https://raoulteeuwen.blogspot.com/2019/08/starting-with-led-strips.html contains some intro video's (after some inspirational ideas for LEDs you see a text "O.k. So even to start with one strip, you need to have a little background." which is followed by some video's with basic knowledge. Can you check whether those are all contain stuff you already know?

NeariX67, 2020-08-28:
https://github.com/Aircoookie/WLED/issues/658

Hmmmm

superbond05, 2020-08-28:
Well that doesn’t help!😩

superbond05, 2020-08-28:
I have double and triple checked everything. Voltage wiring everything thing!
 I first made the mistake by buying $400 worth of the wrong leds. They were the 4 wire with the remote. They working fine just wanted the program my lights and found out I purchased the wrong leds. Now I can’t get them to work at all!
 I need someone to go step buy step from the beginning of what to purchase and how to start right from scratch in layman terms. So I and anyone else that want to get into this hobby will understand. Trust me I have watch YouTube until the caws came home. Still nothing working as should

fishbone-git, 2020-08-28:
What is being suggested by the link to #658 is that this is not the right place for debugging a setup.  This is not likely a "bug", rather a user or setup issue.  You will get more active help in the other places suggested by post #658.  For a starter, I would advise not using a 60A power supply until you have things under control. Far too much can go horribly wrong with that much current and no fusing along the lines.  

Try a stock binary for starters and a short string of LEDs.  You mentioned having the ESP powered and seeing it on the app, THEN powering up your 12V supply, so also good to make sure you have that wiring done correctly, and with the ground of the ESP tied to the GND of the LED power, but not the 5V of the ESP supply and the 5V of the drop-down regulator.

Bright blue probably means you were close, but the wrong colour order for your setup as the default is orange. The multicolour is usually a sign of mis-wiring.  Going dark after a short time: hopefully also a bad connection but can also be something horrible as I suggested could happen at the start of my response. 

Again, start simple and make changes once you have the basic, non-custom first attempt working.

superbond05, 2020-08-29:
Hello
 I figured it out. Alls good thanks

Aircoookie, 2020-08-29:
What was the problem?

closed 2020-08-29 by Aircoookie

superbond05, 2020-08-29:
You sitting down!
I was hooking the data wire to the wrong pin! Lol
It said # 4 on top so I followed the pin from the marking which was the fifth pin! My bad!
Could you answer me 2 more question? I’m putting these sw2811’s on all my windows in the front of my house plus the garage and under the Eaves. I have a 12v 60 amp power supply. I’ll be running 3 wire from window to window and so on. What gauge wire should I use and will I need to add 12v every so often?

Aircoookie, 2020-08-30:
Good to know haha!
With 12v voltage drop generally is less of a problem than with 5 volts and thus you can get away with less power injection. Every 150 to 250 LEDs (3-5 strands) should be enough. As for gauge I am not an expert but if you run multiple cables from the power supply 16 awg should be plenty. If you first run a single cable from the power supply and split it up later, maybe a tad more.
Also I would recommend putting in some car fuses in the distribution if possible since your power supply can output a scary amount of current :)

superbond05, 2020-08-30:
Thank you for the information I really appreciate it. Would you mind if I need more guidance I could contact you?

NeariX67, 2020-08-30:
> 
> Thank you for the information I really appreciate it. Would you mind if I need more guidance I could contact you?
> […](#)

there is a discord community waiting for you...

superbond05, 2020-08-30:
Lol
Not quite understanding you last message

Aircoookie, 2020-08-30:
Discord is the best app for chatting and we use it as the main "home" of the WLED community. Here is a link if you'd like to join the server: https://discord.gg/KuqP7NE 
If it's not for you, don't worry, you can also just post into this thread :)

superbond05, 2020-08-30:
Thanks for being patient with me ( the newbie) I will join.

superbond05, 2020-09-03:
Hello aircoookie
Wondering if you could figure this out for me!
 I have a esp32s chip I flashed it as Dzz discribes in his video. It worked for awhile( while I was cutting and spicing Led strips for around window frames) then it wouldn’t connect to wifi anymore so I tried to reflash it and it would not. I tried holding the boot button and everything else everyone says should work with no avail. So I purchased 2 more esp32’s not the esp32s. They flash with no problem. I can connect to wifi to my laptop. But when I connect it to my power supply I get nothing. I can’t find it in my wifi or wled app. I programmed both and both do the same thing! So that 3 that won’t work. Pretty frustrating. What am I doing wrong?

On Aug 30, 2020, at 6:16 AM, Super <superbond05@hotmail.com> wrote:

 Thank you for the information I really appreciate it. Would you mind if I need more guidance I could contact you?

Aircoookie, 2020-09-03:
Hi!
Try whether it boots/connects if you remove the data connection between pin D4 and the LEDs. Trying to figure out if the power supply or the LEDs are the issue :)