wled/WLED #1120: HTTP API Times out after a while on 0.10.0

fribse, 2020-08-28:
**Describe the bug**
I have placed a WLED/diguno combo inside a sign that we want to use in the IT department here at work.
[image]
https://youtu.be/XInwBBbsWps
We are integrating it into our PRTG survaillance system, to alert of us problems.
But after a while the HTTP API interface gives timeouts. It will come back sometimes, and then go away again.
The WLED is running on a ESP8266 (d1 mini pro).
We just send /win&PL=X as I've predefined the effects (oddly it kept the preloads even after the downgrade).

**To Reproduce**
ESP8266 with wled 0.10.0, and just send requests every 30s in an hour.

**Expected behavior**
Just to change the color, and give 200OK

**WLED version**
 - Board: Wemos D1 mini pro
 - Version 0.10.0
 - Format binary

**Additional context**
I've tried installing 0.8.6 to see if that helps. So far it seems more stable.

Thank you for your help!

fribse, 2020-08-31:
The 0.8.6 also hung during the weekend. The code we made just continuesly set's a preset (every 30 s), it shouldn't hang because of that. If I do a power off/on it wakes up again.

Aircoookie, 2020-08-31:
Hi, sending requests every 30 seconds should indeed be something the module should be able to handle.
Since you also have an external antenna mounted, I assume you have verified that the WiFi signal is consistently strong (in the Info section)
As a test, you could try sending `{"ps":1}` (or a different preset number) to `[WLED-IP]/json`, it uses a different method for request memory allocation. If this would resolve the issue, it would be likely that the problem is a memory leak due to usage of the _String_ class in the HTTP API.

Awesome sign by the way! Translate says it means "Shelters". Could be a replica from a post-nuclear videogame :)

fribse, 2020-08-31:
Hi Aircookie, it is indeed a shelter sign, but it's original, it's around 30 years old :-)
It should have ok coverage, the office is placed rather close to an AP, it shows 48% (-76 db) which is ok. The external antenna was chosen as the sign is made out of metal.
I'll try setting it to use JSON instead.

Aircoookie, 2020-08-31:
Nice!
Hmm, Wifi could indeed still be the culprit here, even though the signal should be quite acceptable. I have an outdoor WLED installation that is reporting `42% (-79 dbm)` right now and it does still go offline at times, even though there is only a single wall between it and the router. None of my inside units (same room as the router) have this issue, seems like reception is generally rather poor with ESPs.

fribse, 2020-08-31:
If it's the wifi it seems odd that a power cycle then fixes it?

chiefymuc, 2020-09-22:
I have the same behaviour. I'm polling the JSON every minute and ever since I updated to 0.10.0.

The module is still connected to the router and MQTT works fine. But when I try to open the UI on the IP adress, it does not load.

Reboot fixes it immediately.

I have multiple devices on 0.10.0. I turned off the polling for one of them, and the problem is gone, I can still access the IP / UI.

fribse, 2020-09-24:
I've updated to 0.10.2 and I've activated the update of the light again, so now it's a waiting game to see if it fails at some point :-)

chiefymuc, 2020-10-18:
Hi @fribse did you have any luck?

I rolled back to 0.9.1 and do not have this issue anymore.

Both 0.10.0 and 0.10.2 have this issue for me.

fribse, 2020-10-20:
I haven't seen it since going to 0.10.2.

chiefymuc, 2020-10-24:
Interestingly, I also saw this issue on 0.9.1 after a longer while.

I flashed 0.10.2 over it and now it has been OK for >24h.

Seems OK for me now.

fribse, 2020-12-24:
Works perfectly with the later updates

closed 2020-12-24 by fribse