wled/WLED #1121: Hyperion Classic und WLED auf RPI4

DaPeace, 2020-08-28:
Ich schreib mal auf Deutsch weil das sicher leichter zu umschreiben ist. Ich habe einen Rpi4 mit Libreelec und Kodi Matrix. Das ist eine Milhouse-Testversion. Ich hab hier hyperion installiert und das schickt die Daten per UDP and WLED. Leider bekomme ich hier zwar in WLED angezeigt das Daten kommen aber die LEDs sind aus.
Kann ich die UDP-Schnittstelle irgendwie manuell austesten und schauen ob woran das liegt? Bis jetzt find ich das WLED echt super. Jetzt müsste ich das nur noch mit dem Grabber vom Kodi verheiraten können..

Danke schonmal, Grüsse, DaPeace

sansillusion, 2020-08-28:
Hi I have used google translate to read your message. If your wled is working fine but when you activate Hyperion it turns dark it may be your gamma correction in Hyperion setup. I had a similar issue that when screen was not fully white the LEDs would be very dark of off, when I changed gamma correction in Hyperion config it ran perfectly. My suggestion is to test the wled without Hyperion, if it's working normally then your issue is most likely with Hyperion config.

DaPeace, 2020-08-29:
Hello and thank you for your reply. My idea was to first check if the udp-connection is working and wled is showing a light by sending a valid udp-line to wled. wled is working with its build in effects and is showing that data is coming from hyperion but i cant see if its valid.

sansillusion, 2020-08-29:
So I believe that you are almost there. If wled does get show that Hyperion is connected and works normally without hyperion then it must be an Hyperion configuration issue. I aslo use Kodi on a Rpi4. here is a link to my config might help you spotting some setting. https://drive.google.com/file/d/1-CLCrWUA3YqCmEQdShE6PypU9UB8zYgr/view?usp=sharing

PaulNBerlin, 2020-09-04:
I had the same issue with the latest hyperion version. Seems there is a bug. Try installing an older version. That worked for me.

closed 2021-01-09 by stale[bot]