wled/WLED #1122: Display scrolling text?

Bihel29, 2020-08-30:
Hello, I have a question: 
Is there a way to display scrolling text? 

I made a 3D lamp with 200 pixels and use WLED on it.

I would like to display the text like with Glediator / Jinx but using WLED is this possible? 

My lamp: https://i.imgur.com/4S8z2bH.jpg
I would like : https://i.imgur.com/drexllE.jpg

Thanks

Aircoookie, 2020-08-30:
Hi!
Yes, using WLED with Jinx is supported! I would recommend the TPM2.net protocol supported in WLED master (not yet in the latest release 0.10.0, but there is a new release 0.10.2 coming tomorrow if you are using the binary) as it is somehow difficult to get more than 170 LEDs to work via E1.31 in Jinx.

Bihel29, 2020-08-30:
Hi , thanks for your answer.

I directly use the source to upload WLED on my Wemos so I am in 0.10.2?

In Jinx, I configured my Matrix Options, then in output Devices I configured the IP address of my Wemos with the TPM2.NET protocol.

I run a test with Jinx and nothing happens, what should I do?

I'ts the first time with jinx so i am a bit confused.

Sorry for my bad english

Thanks

Aircoookie, 2020-08-30:
If you are using the latest source, support is in, awesome!
One important thing I forgot to mention is that you need to set the WLED broadcast port in sync settings to 65506 and reboot once! Hope that solves it :)

Bihel29, 2020-08-30:
It works, thank you :)

closed 2020-08-30 by Aircoookie

pattyland, 2020-09-09:
Hi, I want to do the same but I can't get jinx to change any of me leds. Controlling via the web interface is working great.
I set the port to 665506 and rebooted, I added the IP of my wled esp with tpm2.net driver and configured the output page. 
When I click on "start output" nothing happens :(￼￼￼

micw, 2022-11-13:
Hello,

I found this ticket by searching if it is possible to show text with WLED. Would it be possible without an external software, directly with WLED?

blazoncek, 2022-11-13:
Yes, change segment name.
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.